opencomputeproject/onie #683: ONIE install/update timestamp in onie-sysinfo command

psarker, 2018-01-19:
It would have been really nice if we can display ONIE installation timestamp as the part of onie-sysinfo command.
e.g. We can use onie-sysinfo -T (or whatever more appropriate).

cbrune, 2018-01-19:
Indeed, include this information in a few place.  `onie-sysinfo` for sure, but I would also like to see it in the `onie-version` command that a NOS can run from /mnt/onie-boot/onie/tools/bin.

psarker, 2018-01-22:
it would be really great to see the installation time in onie-version file.. picture

cbrune, 2018-01-22:
Something to consider about `install time`.  The system real time clock (RTC) may not be set correctly at the time ONIE is installed.  That would lead to ONIE recording bogus install times.

So one must take the "install time" with a grain of salt, i.e. it is informational only, do not rely on it being 100% correct.

A possible mitigation is for ONIE to run `ntpd` and update the system time at boot.  Busybox does include a ntp client, so technically it is not that hard.  Also we can request NTP servers via DHCP.  The bigger problem will be operationally, where the end customer's DHPC infrastructure needs to be configured to send out NTP server.   That may or may not be setup.

All of this comes back to this: take the "install time" with a grain of salt.