opencomputeproject/onie #687: Have a REST API

cbrune, 2018-01-29:
For end users, debugging why ONIE is not downloading their image can be challenging.  Also using ssh to access the box can also be challenging.

The most common scenario for ssh usage is seeing logs... Second is manually kicking the install because DHCP is not owned by network team.

The proposal here is to add a REST API for common ONIE operations, decreasing the need for end users to ssh to the box.  Specific useful operations needed:

- get logs
- get eeprom contents
- get platform informat
- POST NOS install URL
- POST FW update URL

Using a POST to launch a NOS install could be quite a bit easier than using ssh and typing `onie-nos-install`.

basilsaji-rtb, 2018-02-06:
I absolutely concur with this thought. It would actually be even better if ONIE can support the initial provisioning using the REST model.

The REST url is provided by the DHCP http server option. ONIE then posts an http request with all the parameters like serial number, mac address, etc to the REST url and expects the response(in a specified format) that will contain the location of the image url to install.

cbrune, 2018-03-19:
@basilsaji-rtb - appreciate your thoughts.  Let me see if I get this straight:

- ONIE makes a DHCP request
- DHCP server responds with a REST URL
- ONIE posts variable parameters to the REST URL
- The REST API replies with another URL of the actual image to install

Is that basically it?

That puts a bit of responsibility on the end user to run a HTTP server than can reply to the REST API.  Not impossible, but a little tough.

ONIE already has a similar, though not REST, mechanism where:

- ONIE makes a DHCP request
- DHCP server responds with an image URL
- ONIE sets *HTTP header* during the GET of the image URL
- The HTTP server can inspect the HTTP header parameters and send back the correct image

This is described in more detail here:
https://opencomputeproject.github.io/onie/user-guide/index.html#http-headers-and-scripting
https://opencomputeproject.github.io/onie/design-spec/discovery.html#http-headers

pichuang, 2018-03-20:
The RESTful API "GET" is fine for me.
Regarding POST NOS/FW, is it any security issue if no authority process?