opencomputeproject/onie #688: Boot ONIE Using Secure Boot

cbrune, 2018-01-29:
This goal has been stated numerous times.  Here is a good reference:

http://lists.opencompute.org/pipermail/opencompute-onie/2017-April/001388.html

The main ideas:

1. ONIE should boot with UEFI Secure Boot enabled.
2. ONIE should verify that NOS installer images pass certain security checks

kielfriedt, 2021-03-17:
I see mention of UEFI on x86, with ARM64 now using UEFI does this plan on being pushed to arm platforms?

ehdoyle, 2021-08-03:
Well, secure boot is finally in now in for x86.  ARM support is pending getting the arm emulation running, and there's some talk about doing an installer that would get verified by the keys in the switch's BIOS...