opencomputeproject/onie #690: GRUB menu: make installed NOS the top entry

cbrune, 2018-01-29:
`install mode` is currently the default GRUB menu item when entering the ONIE GRUB.

This can be a surprise for people who are manually navigating the GRUB menus from their NOS and all of a sudden find themselves in ONIE's GRUB and booting into install mode.  From there people get burned since install mode is sticky and now their NOS is gone.

For some discussion see: http://lists.opencompute.org/pipermail/opencompute-onie/2018-January/001617.html

The proposal here is to investigate the possibility of having a NOS be the *default* ONIE GRUB menu entry.  Specifically the behavior falls into two separate scenarios.

Scenario 1 -- Completely fresh system that only has ONIE installed.  No NOS yet.  In this case the default ONIE GRUB entry should be `install mode`.  Also install mode continues to remain sticky. The menu looks like:
```
   +----------------------------------+
   |*ONIE: Install OS                 |  <-- default
   | ONIE: Rescue                     |
   | ONIE: Uninstall OS               |
   | ONIE: Update ONIE                |
   | ONIE: Embed ONIE                 |
   | DIAG: Vendor Diagnostic          |
   +----------------------------------+
```

Scenario 2 -- A NOS is installed.  The user accidentally enters the ONIE GRUB menus.  In this case the default ONIE GRUB menu item would be for the NOS.  The menu would look like:
```
   +----------------------------------+
   |*NOS: Network Operating System    |  <-- default
   | ONIE: Install OS                 |
   | ONIE: Rescue                     |
   | ONIE: Uninstall OS               |
   | ONIE: Update ONIE                |
   | ONIE: Embed ONIE                 |
   | DIAG: Vendor Diagnostic          |
   +----------------------------------+
```
In this scenario if the end user stumbles into the ONIE GRUB menus they will simply boot back into their NOS.

Scenario 3 -- The user requests `install mode` on purpose, knowing that it is sticky.  As part of install mode, ONIE will make itself the new default.  The GRUB menu will now look like this (and remain this way until a NOS is installed):
```
   +----------------------------------+
   |*ONIE: Install OS                 |  <-- default
   | NOS: Network Operating System    |
   | ONIE: Rescue                     |
   | ONIE: Uninstall OS               |
   | ONIE: Update ONIE                |
   | ONIE: Embed ONIE                 |
   | DIAG: Vendor Diagnostic          |
   +----------------------------------+
```

In this scenario the user can still decide to boot back into their NOS.  Using [onie-boot-mode](https://opencomputeproject.github.io/onie/cli/index.html#onie-boot-mode) from their NOS, the end user could clear the `install mode` and put the NOS entry back as the default.

All of this change can be effected for x86 systems by updating the grub.d fragment that ONIE provides for NOSes to use:   https://github.com/opencomputeproject/onie/blob/master/installer/grub-arch/grub.d/50_onie_grub

Also the `onie-boot-mode` may need some enhancement to make all this work.

lakesh-sharma, 2018-02-26:
Making rescue mode as default selection can be a bit simplified solution for this issue.
When unit ships from MFG i.e. after diags and onie present, then Install-OS mode can be set as default mode, but the menu would remain same as shown below (except a diag entry at end).

After nos install, the nos-installer can again set rescue mode a default (optional).
If some one really want to boot to install mode, he/she can set it from onie/NOS, and being persistent,
it would stay there, even if menu appearance would still show rescue as first as shown below.
Just a thought.

 |* ONIE: Rescue                |  <-- default
   | ONIE: Install OS                    |
   | ONIE: Uninstall OS               |
   | ONIE: Update ONIE                |
   | ONIE: Embed ONIE

cbrune, 2018-02-27:
I see your point.  What we *really* need is for install mode to be the default when it leaves the factory.  Also it remains the default (is sticky), until explicitly switched.

A NOS installer could permanently make the default RESCUE mode.  As you say that would be optional for the NOS to do.

So the trick for ONIE is to offer a method for the NOS installer to set RESCUE mode as the default.  It seems we could have a little state machine using a couple of GRUB env variables and make this happen.

commit 40b7011f (2018-03-13): u-boot: manage NOS mode on successful install

This patch implements the NOS mode behavior for U-Boot based
architectures.

After a successful NOS install the onie_nos_mode variable is set to
'yes'.

When onie_nos_mode is 'yes', manually entering install mode is no
longer a one-way trap door.

Closes: #690
commit cf6e91cd (2018-03-13): NOS mode: a persistent NOS mode boot option

This patch adds the notion of "NOS installed" mode.  This persistent
state is set to true after a successful NOS install.

This patch changes the following behavior for all CPU architectures:

1. Upon a successful NOS install, the system sets a persistent
   environment variable, onie_nos_mode, to "yes".

2. When onie_nos_mode is 'yes', entering install mode is no longer
   sticky. I.e. entering install mode is no longer a one-way trap
   door.

Furthermore, for GRUB architectures, the default GRUB menu item now
defaults to "rescue" mode, instead of [... cut]
closed 2018-03-13 by cbrune