opencomputeproject/onie #697: Enhancement request: DHCP option 61 support

basilsaji-rtb, 2018-02-06:
We have a requirement where we want ONIE's dhcp request to contain option 61(with serial number of the device) so that we can use the serial number in the configuration of the dhcp server for provisioning the NOS images instead of the mac address that is currently available. Kindly help with the same.

cbrune, 2018-02-06:
DHCP requests are made by the busybox `udhcpc` client:
```
ONIE:/ # udhcpc --help                                                                                                       
BusyBox v1.25.1 (2018-01-30 16:43:57 PST) multi-call binary.                                                                 
                                                                                                                             
Usage: udhcpc [-fbqvRB] [-a[MSEC]] [-t N] [-T SEC] [-A SEC/-n]                                                               
        [-i IFACE] [-s PROG] [-p PIDFILE]                                                                                    
        [-oC] [-r IP] [-V VENDOR] [-F NAME] [-x OPT:VAL]... [-O OPT]...                                                      
                                                                                                                             
        -i,--interface IFACE    Interface to use (default eth0)                                                              
        -s,--script PROG        Run PROG at DHCP events (default /usr/share/udhcpc/default.script)                           
        -p,--pidfile FILE       Create pidfile                                                                               
        -B,--broadcast          Request broadcast replies                                                                    
        -t,--retries N          Send up to N discover packets (default 3)                                                    
        -T,--timeout SEC        Pause between packets (default 3)                                                            
        -A,--tryagain SEC       Wait if lease is not obtained (default 20)                                                   
        -n,--now                Exit if lease is not obtained                                                                
        -q,--quit               Exit after obtaining lease                                                                   
        -R,--release            Release IP on exit                                                                           
        -f,--foreground         Run in foreground                                                                            
        -b,--background         Background if lease is not obtained                                                          
        -S,--syslog             Log to syslog too                                                                            
        -a[MSEC],--arping[=MSEC] Validate offered address with ARP ping                                                      
        -r,--request IP         Request this IP address                                                                      
        -o,--no-default-options Don't request any options (unless -O is given)                                               
        -O,--request-option OPT Request option OPT from server (cumulative)                                                  
        -x OPT:VAL              Include option OPT in sent packets (cumulative)                                              
                                Examples of string, numeric, and hex byte opts:                                              
                                -x hostname:bbox - option 12                                                                 
                                -x lease:3600 - option 51 (lease time)                                                       
                                -x 0x3d:0100BEEFC0FFEE - option 61 (client id)                                               
        -F,--fqdn NAME          Ask server to update DNS mapping for NAME                                                    
        -V,--vendorclass VENDOR Vendor identifier (default 'udhcp VERSION')                                                  
        -C,--clientid-none      Don't send MAC as client identifier                                                          
        -v                      Verbose                                                                                      
Signals:                                                                                                                     
        USR1    Renew lease                                                                                                  
        USR2    Release lease                                                                                                
```

The relevant option here is `-x OPT:VAL`.  The help output above even has an example for option 61, it is in HEX.

ONIE has a function dedicated to generating all the options for `udhcpc`, which is here:
https://github.com/opencomputeproject/onie/blob/master/rootconf/default/lib/onie/functions#L395-L412

The trick would be to add `-x 0x3d:........` to the option list.  There is already a helper function `str2hex()` to convert a string into a hex values.  So all the pieces are there.

Another question is what value to put in the one-byte "type field" of option 61?  See https://www.ietf.org/rfc/rfc2132.txt section 9.14.

idefixcert, 2019-11-27:
@cbrune could you help with implementing that?

closed 2020-03-18 by ehdoyle

Vincent-Chang-Ufi, 2024-12-13:
Hi I am wondering that if ONIE has multiple interface (ex : eth0,eth1) , and both request the ip from DHCP server through same serial number. eth0 and eth1 will get the same ip, it exist the conflict. Anyway can workaround ? Please see the attached Picture below :

[image]

commit 7667fb6b (2020-03-18): Add client ID (option 61) to udhcp parameters.

Sends the serial number of the device in the DHCP packet as option 61,
formatted as a hex converted ASCII string
(Client ID).

Closes: #697