opencomputeproject/onie #708: Kexec error on usb install

ecrespi79, 2018-03-07:
After making the making the Debian image.  I loaded the file onto usb and it is detected however I receive kexec error when loading the new kernel.  Is this an error with onie or the debian-jessie-amd64-mini-ONIE.bin that was created? 

I followed the https://github.com/opencomputeproject/onie/tree/master/contrib/debian-iso to make the bin.
 
Mar  7 19:45:38 onie local0.info discover: ONIE: Starting ONIE Service Discovery
Mar  7 19:45:54 onie local0.info discover: onie_neighs =
Mar  7 19:45:54 onie local0.info discover: onie_local_parts = /dev/sdb1,
Mar  7 19:45:55 onie local0.info onie-exec: ONIE: Executing installer: file://dev/sdb1/onie-installer
Mar  7 19:45:56 onie local0.info os-install: Verifying image checksum ... OK.
Mar  7 19:45:56 onie local0.info os-install: Preparing image archive ... OK.
Mar  7 19:45:56 onie local0.info os-install: Loading new kernel ...
Mar  7 19:45:56 onie local0.info os-install: /tmp/tmp.ktZ205/installer/install.sh: line 18: kexec: not found

ecrespi79, 2018-03-07:
I was able to load a cumulus image that I placed on the same usb stick. This did not throw the kexec error. 

ONIE: Starting ONIE Service Discovery
ONIE: Executing installer: file://dev/sdb1/onie-installer-x86_64
Verifying image checksum ...OK.
Verifying image architecture ...OK.
Verifying system ram ...OK.
Preparing image archive ... OK.
Setting up installer ...45258 blocks
Info: The full install log is located at /tmp/ei_runlog.XXGScOH5
 OK.
Please reboot to start installing OS.
Rebooting...

cbrune, 2018-03-07:
What version of ONIE are you running?  Type `onie-sysinfo -a` and `cat /etc/machine.conf`.

The kexec feature first appeared in ONIE version 2015.08.

ecrespi79, 2018-03-07:
It looks like the version we have is a little older.  I will work on getting that updated.  Thanks for the info. 

Loader Version 0x29 13 2015.05.0.0.2