opencomputeproject/onie #712: Did ONIE support install more one NOS？

lockless, 2018-03-16:
tow questions：
1. Can i install two ONIE on my device for backup？ 
    When i install nos on one of the onie， and finally break the onie system， i can easily change to another one。

2. Can i installer two or more NOS on the storage？
    When one of the nos failed， i can easily change to some other instead of reinstall。

think for advise！

cbrune, 2018-03-16:
Hello @lockless - thanks for the questions.

> Can i install two ONIE on my device for backup？
When i install nos on one of the onie， and finally break the onie system， i can easily change to another one。

Currently there is no support for having two ONIE's installed on a system.  Presumably this means installing ONIE on two separate disks within the system.  Some changes would be required to figure out which ONIE is the "active" ONIE.

> Can i installer two or more NOS on the storage？
When one of the nos failed， i can easily change to some other instead of reinstall。

This can be done today, though it depends on the particular NOS installer.  Some NOS installers wipe out everything, except ONIE partitions, as part of the install process.  A NOS installer would have to know (or be guided) to leave an existing NOS installed, while installing an additional NOS.

lakesh-sharma, 2018-03-16:
moving to uefi arch. based onie, which is already supported, would solve this issue.
Being uefi, the nos itself or its installation, can not brink the system. User would always have an option to
debug and boot using available OS(s), including onie,diags, and nos.

lockless, 2018-03-19:
thanks for answer ! I will try to experiment this.

lockless, 2018-03-22:
I try to install to SONiC OS on my device， when install the second times， the log as follow：
```
onie_platform: x86_64-bf002-r0
deleting partition 3 ...
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot.
The operation has completed successfully.
Partition #1 is in use.
Partition #2 is in use.
Partition #3 is available
Creating new SONiC-OS partition /dev/sda3 ... 
```

as i konw that, first time i install it, the partition should be:
```
Partition 1:  /dev/sda1 ESP                  ==> use for EFI
Partition 2:  /dev/sda2 ONIE-BOOT    ==>for ONIE OS
Partition 3:  /dev/sda3 SONiC-OS       ==>for SONiC OS
```

now, when install SONiC the second time, i expect that it create partition 4, and install SONiC-OS on partition, but it delete partition 3 first.

for my porpose,  what can i do?

thanks for advise. @cbrune @lakesh-sharma

lockless, 2018-03-22:
```
deleting partition 3 ...
```
is this check and control by ONIE or SONiC installer?
which one should i try to fix?

cbrune, 2018-03-22:
Hi @lockless  -- that is controlled by the SONIC installer.

lockless, 2018-03-23:
thanks， it works now！

closed 2018-03-23 by lockless