opencomputeproject/onie #724: What is the compiler environment for onie?

fansss1116, 2018-04-02:
I have compiled onie for some time, and the compilation environment is centos6.5.
A lot of errors occurred during compilation.
I would like to ask what is onie's standard compilation environment under centos?
thanks

bluejumper, 2018-04-03:
I suggest using Docker to create a build environment.

I've attached a possible solution to this link: [A Debian build environment.](https://github.com/bluejumper/build-onie)

Please note that this image remains untested; I created this after creating my own environment manually. I hope it serves as a starting point.

ONIE will build :)

cbrune, 2018-04-04:
@bluejumper - that is really cool.  I haven't played with docker much before, but this got me going.

I think this a good way to go for folks who having problems setting up a native environment.

I have a pull request head your way with a few updates.

cbrune, 2018-04-04:
@fansss1116 -- the standard environment for ONIE is Debian 9.   See the `debian-prepare-build-host` target in the Makefile for the necessary packages on the host environment.  Those do not change very often, but they do change as new features are added to ONIE.  You may have some missing packages.

Also see this wiki page:  https://opencomputeproject.github.io/onie/developers/building.html

cbrune, 2018-04-04:
@bluejumper - see this: https://github.com/bluejumper/build-onie/pull/1

fansss1116, 2018-04-04:
@cbrune Thank you. I'll try it.

PR #728 (2018-04-05): docker-build: create ONIE build environment
Create a docker container to use as an ONIE build environment.

See the README and Dockerfile for details.

Big thanks to Benayak Karki for providing the initial implementation
upon which this code based.

Closes: #724 #725
commit 9fa62dd4 (2018-04-05): docker-build: create ONIE build environment

Create a docker container to use as an ONIE build environment.

See the README and Dockerfile for details.

Big thanks to Benayak Karki for providing the initial implementation
upon which this code based.

Closes: #724 #725
closed 2018-04-05 by cbrune