opencomputeproject/onie #731: onie-syseeprom: the --set argument parser is case sensitive

cbrune, 2018-04-12:
This is a dumb one.

Setting an EEPROM TLV value using an *upper* case hex value fails, while using a *lower* case value works.

This fails:
```
onie-syseeprom --set 0x2B='Hardware Boxes R Us'
```

Instead the `2B` needs to be lower case: `2b`.  This works:

```
onie-syseeprom --set 0x2b='Hardware Boxes R Us'
```

closed 2018-04-13 by cbrune

scott-emery, 2018-04-13:
Thank you for fixing this @cbrune! This has always annoyed me when (re)programming an EEPROM. I could never remember if the lowercase or uppercase hex digits were the ones accepted.

cbrune, 2018-04-13:
Interestingly enough I discovered that *nginx* httpd works OK without brackets.  This is the httpd we use with the ONIE compatibility tests.  That is why this bug hadn't been seen.

I definitely see the problem when using *apache2* httpd.

PR #732 (2018-04-13): onie-syseeprom: accept mixed case TLV hex codes
Previously, the onie-syseeprom "--set" argument only accepted lower case TLV hex code.  This patch accepts both lower and upper case hex codes.

Closes: #731
commit 9aa85e56 (2018-04-13): onie-syseeprom: accept mixed case TLV hex codes

Previously, the onie-syseeprom "--set" argument only accepted lower
case TLV hex code.  This patch accepts both lower and upper case hex
codes.

Closes: #731