opencomputeproject/onie #735: Instructions for creating x86_64 KVM ONIE iso fails

q2dg, 2018-04-16:
_make -j4 MACHINE=kvm_x86_64 all_ command (described in https://github.com/opencomputeproject/onie/blob/master/machine/kvm_x86_64/INSTALL) doesn't show any error, but resulting files aren't generated. I'm using Fedora 27 with a 4.15.16-300 kernel

PD: I've installed all necessary packages, in special "stgit" and "gperf" besides the usual suspects (git, gcc, make...)

cbrune, 2018-04-17:
Hi @q2dg -

A couple of things.

1.  if anything went wrong during your first attempts, like a necessary package was missing and the xtool build failed, then you need to do `make distclean` first and then do `make -j4 MACHINE=kvm_x86_64 all` to start fresh.  There's a bug in the xtool makefile that needs fixing.

2. the build is only officially supported for Debian 9 systems.  For other distros trying building in a Debian 9 docker image, described here:  https://github.com/opencomputeproject/onie/tree/master/contrib/build-env

q2dg, 2018-04-18:
Thanks a lot for the explanation, I didn't know about point 1, thanks. 
I've tried again but with no luck, though.
Anyway, as my system is Fedora, I'll close the issue, if you don't mind.

closed 2018-04-18 by q2dg