opencomputeproject/onie #738: I install SONiC ONIE Image on my supermicro SSE-G3648B and get some error

wjf1993, 2018-04-20:
hello everyone.I follow the steps of the document to install(https://github.com/Azure/SONiC/wiki/Quick-Start) and I finished installing the NOS.But i don't why i can only “welcome grub”.so How should I go back to the ONIE environment？

cbrune, 2018-04-20:
Hi @wjf1993 

You will have to ask the sonic folks how to get back into ONIE from their OS.  How that is done is OS specific.

For Cumulus Linux, as root run `onie-select -r` followed by a reboot.

wjf1993, 2018-04-23:
Thank you @cbrune 
Because I don't know where is the error in my device,so I submitted issue here.Thank you very much for your advice and i will try it.

cbrune, 2018-04-23:
Hi @wjf1993  -

You try submitting an issue to their project:  https://github.com/Azure/SONiC/issues

Or their mailing list: https://groups.google.com/forum/#!forum/sonicproject

closed 2018-04-24 by wjf1993