opencomputeproject/onie #741: How is ONIE installed on a blank hard disk? What I saw in the case is rather sketchy. Is there a complete tutorial?

wjf1993, 2018-04-24:


cbrune, 2018-04-24:
Depends on the hardware platform.   First you need to know is it PowerPC, ARM or x86?  For x86 you also need to know whether it is using legacy BIOS or UEFI.

cbrune, 2018-04-24:
For x86 the nuts and bolts are here:
https://opencomputeproject.github.io/onie/design-spec/x86_arch_design.html

For PowerPC/ARM the info is here:
https://opencomputeproject.github.io/onie/design-spec/uboot_arch_design.html

wjf1993, 2018-04-25:
@cbrune Thank you and sorry. Maybe my engilsh is not good enough  to describe this issue correctly.I mean  I want to install ONIE on a blank hard disk(e.g. USB).It just a disk like install a window PE sys on USB.
(**My device is x86 and i want it use UEFI to boot ONIE)

cbrune, 2018-04-25:
ONIE is designed to support a specific hardware platform and to be installed permanently on a hard disk.  It is not intended to be installed on removable media, like a USB.

What hardware platform are you targeting?

wjf1993, 2018-04-26:
My device is SUPEMICRO SSE-G3648B/Intel Atom Rangeley (C2338) CPU Module.Now i try to use machine/dell/dell_s4048t_c2338 imge to install ONIE on this switch.But i get some error.
Checking for changes in the working directory ... Importing patch "feature-onie-syseeprom-sysfs.patch" ... done
==== Extracting upstream grub ====
Importing patch "gptfdisk-onie-cross-compile.patch" ... done
Importing patch "onie-sys-eeprom-platform.patch" ... error: patch failed: miscutils/sys_eeprom.c:68
error: miscutils/sys_eeprom.c: patch does not apply
done
Importing patch "add-onie-partition-types.patch" ... done
Now at patch "add-onie-partition-types.patch"
stg import: Diff does not apply cleanly
make/busybox.make:74: recipe for target '/home/wjfroot/onie_test/onie/build/dellemc_s4100_c2338-r0/stamp/busybox-patch' failed
make: *** [/home/wjfroot/onie_test/onie/build/dellemc_s4100_c2338-r0/stamp/busybox-patch] Error 2
make: *** Waiting for unfinished jobs....

it look like The file path is incorrect.

wjf1993, 2018-04-26:
My device is SUPEMICRO SSE-G3648B/Intel Atom Rangeley (C2338) CPU Module.Now i try to use machine/dell/dell_s4048t_c2338 imge to install ONIE on this switch.But i get some error.
Checking for changes in the working directory ... Importing patch "feature-onie-syseeprom-sysfs.patch" ... done
==== Extracting upstream grub ====
Importing patch "gptfdisk-onie-cross-compile.patch" ... done
Importing patch "onie-sys-eeprom-platform.patch" ... error: patch failed: miscutils/sys_eeprom.c:68
error: miscutils/sys_eeprom.c: patch does not apply
done
Importing patch "add-onie-partition-types.patch" ... done
Now at patch "add-onie-partition-types.patch"
stg import: Diff does not apply cleanly
make/busybox.make:74: recipe for target '/home/wjfroot/onie_test/onie/build/dellemc_s4100_c2338-r0/stamp/busybox-patch' failed
make: *** [/home/wjfroot/onie_test/onie/build/dellemc_s4100_c2338-r0/stamp/busybox-patch] Error 2
make: *** Waiting for unfinished jobs....

This seems to be caused by the incorrect path of the file.

wjf1993, 2018-04-26:
My device is SUPEMICRO SSE-G3648B/Intel Atom Rangeley (C2338) CPU Module.Now i try to use machine/dell/dell_s4048t_c2338 imge to install ONIE on this switch.But i get some error.
Checking for changes in the working directory ... Importing patch "feature-onie-syseeprom-sysfs.patch" ... done
==== Extracting upstream grub ====
Importing patch "gptfdisk-onie-cross-compile.patch" ... done
Importing patch "onie-sys-eeprom-platform.patch" ... error: patch failed: miscutils/sys_eeprom.c:68
error: miscutils/sys_eeprom.c: patch does not apply
done
Importing patch "add-onie-partition-types.patch" ... done
Now at patch "add-onie-partition-types.patch"
stg import: Diff does not apply cleanly
make/busybox.make:74: recipe for target '/home/wjfroot/onie_test/onie/build/dellemc_s4100_c2338-r0/stamp/busybox-patch' failed
make: *** [/home/wjfroot/onie_test/onie/build/dellemc_s4100_c2338-r0/stamp/busybox-patch] Error 2
make: *** Waiting for unfinished jobs....

It looks like a patch with a conflict.

cbrune, 2018-04-26:
A fix for that eeprom patch issue went into the repo yesterday.  see 6b31dcf780d2d26c8b7fe8abc420e9b3516a2b88

update your tree and try again.  I would first do this:
```
make -j4 MACHINEROOT=../machine/dell MACHINE=dell_s4048t_c2338 user-clean
```

and then:

```
make -j4 MACHINEROOT=../machine/dell MACHINE=dell_s4048t_c2338 all
```

wjf1993, 2018-04-27:
@cbrune With your help, my problems have been solved. Thank you very much.Now i try to install ONIE to my device.

Thank you!
O(∩_∩)O

closed 2018-05-03 by wjf1993