opencomputeproject/onie #751: base MAC address

lockless, 2018-05-21:
As says 
```
The default ONIE behaviour is to assign MAC addresses to the Ethernet
management interfaces sequentially from the board EEPROM's base MAC
address.
```

but it seems that the eth0's mac address did not use what i set in eeprom.

so, which code has been called,
or, what source code can i read the get this information ?

thanks for advise!!

cbrune, 2018-05-21:
The relevant code is here:
https://github.com/opencomputeproject/onie/blob/master/rootconf/default/etc/init.d/networking.sh#L202-L212

The `$onie_skip_ethmgmt_macs` variable is controlled by a Makefile variable `SKIP_ETHMGMT_MACS`, which defaults to `no`.

That said, it is really better use the burned in MAC address of eth0 if it is valid.  The Intel NIC on most x86_64 platforms have a little EEPROM associated with them that the Intel NIC driver knows how to read.  If that little EEPROM (not the board EEPROM) has a valid MAC address I recommend to use that instead of assigning eth0 a MAC from the board EEPROM.  It makes everyone's life easier to use the burned in MAC address.  If needed for manufacturing you can google around find the `ethtool` incantations used to program that little EEPROM from the Linux command line.

In any event, I recommend setting `SKIP_ETHMGMT_MACS` to `yes` if possible.

The reasons this is better goes as follows.

The machine first boots up in ONIE mode and does an image discovery.  Using the burned in eth0 MAC address, the system will use DHCP to obtain all the network parameters.  At a customer site, a network engineer modified the DHCP server configuration to match that MAC address.

Next, the the OS is installed and the OS boots.  Now, the problem for the OS is whether or not eth0 is using the burned in MAC or the first MAC from the board EEPROM.  If the OS chooses wrong, then DHCP will be out of wack, making the network engineers grumpy.  It is better to just rely on the burned in eth0 MAC in all cases if possible.  That is how normal server hardware works.

The default you quote above about using the first board EEPROM MAC is a bit out of date -- That approach originated when ONIE was primarily used on PowerPC hardware, where the NICs did not have a little EEPROM for holding its own MAC address.  Back then all the data needed to be stored in the board EEPROM.

I'll make a note to update the ONIE documentation to that effect here.

cbrune, 2018-05-21:
@lockless  - where did you get that quoted text from?  I'm looking at the wiki documentation and cannot find.  Wherever it is, I would like to edit and clarify it.