opencomputeproject/onie #753: build process for kvm_x86_64 has errors preventing successful build

jamesbeedy, 2018-05-24:
Here is the [output](https://paste.ubuntu.com/p/69BKgQWNXF/) of building onie. Seems there are a few errors, possibly I'm doing something wrong?

cbrune, 2018-05-24:
Hi @jamesbeedy -

It looks like this is a re-start of a previously failed build attempt (lines 3-5 of your log file).

There's a known bug that re-starting the build of the cross-compiler does not work right: see #736

To proceed first use `make distclean` and then try your build again.

General suggestion -- Line 1 from log file shows the use of `sudo`.  Please do *not* use `sudo` when building ONIE.  It is unnecessary and will likely cause problems.

If you still have problems building natively, try building using the build environment container.  That process is described here:  
https://github.com/opencomputeproject/onie/tree/master/contrib/build-env

jamesbeedy, 2018-05-24:
@cbrune thanks for the response here.

Cleaning the env with `make distclean` and running without sudo got my build much further.

Now my build fails at:
```bash
Makefile:35982: recipe for target 'normal_module-grub_script.yy.o' failed
make[4]: *** [normal_module-grub_script.yy.o] Error 1
make[4]: Leaving directory '/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub-host/grub-x86_64-efi/grub-core'
[INFO ]  Extracting and patching toolchain components: done in 95.19s (at 01:39)
Makefile:23635: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub-host/grub-x86_64-efi/grub-core'
Makefile:10901: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub-host/grub-x86_64-efi'
Makefile:3127: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub-host/grub-x86_64-efi'
make/grub.make:333: recipe for target '/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/stamp/grub-host-build-x86_64-efi' failed
make: *** [/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/stamp/grub-host-build-x86_64-efi] Error 2
make: *** Waiting for unfinished jobs....
gcc -DHAVE_CONFIG_H -I. -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/grub-core -I..  -Wall -W  -DGRUB_MACHINE_PCBIOS=1 -DGRUB_MACHINE=I386_PC -m32 -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/include -I../include -DGRUB_FILE=\"term/gfxterm.c\" -I. -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/grub-core -I.. -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02 -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/include -I../include -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/grub-core/lib/libgcrypt-grub/src/    -D_FILE_OFFSET_BITS=64 -Os -Wall -W -Wshadow -Wpointer-arith -Wundef -Wchar-subscripts -Wcomment -Wdeprecated-declarations -Wdisabled-optimization -Wdiv-by-zero -Wfloat-equal -Wformat-extra-args -Wformat-security -Wformat-y2k -Wimplicit -Wimplicit-function-declaration -Wimplicit-int -Wmain -Wmissing-braces -Wmissing-format-attribute -Wmultichar -Wparentheses -Wreturn-type -Wsequence-point -Wshadow -Wsign-compare -Wswitch -Wtrigraphs -Wunknown-pragmas -Wunused -Wunused-function -Wunused-label -Wunused-parameter -Wunused-value  -Wunused-variable -Wwrite-strings -Wnested-externs -Wstrict-prototypes -g -Wredundant-decls -Wmissing-prototypes -Wmissing-declarations  -Wextra -Wattributes -Wendif-labels -Winit-self -Wint-to-pointer-cast -Winvalid-pch -Wmissing-field-initializers -Wnonnull -Woverflow -Wvla -Wpointer-to-int-cast -Wstrict-aliasing -Wvariadic-macros -Wvolatile-register-var -Wpointer-sign -Wmissing-include-dirs -Wmissing-prototypes -Wmissing-declarations -Wformat=2 -march=i386 -m32 -mrtd -mregparm=3 -falign-jumps=1 -falign-loops=1 -falign-functions=1 -freg-struct-return -mno-mmx -mno-sse -mno-sse2 -mno-sse3 -mno-3dnow -msoft-float -fno-dwarf2-cfi-asm -mno-stack-arg-probe -fno-asynchronous-unwind-tables -fno-unwind-tables -Qn -fno-PIE -fno-pie -fno-stack-protector -Wtrampolines -Werror   -ffreestanding   -MT term/gfxterm_module-gfxterm.o -MD -MP -MF term/.deps-core/gfxterm_module-gfxterm.Tpo -c -o term/gfxterm_module-gfxterm.o `test -f 'term/gfxterm.c' || echo '/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/grub-core/'`term/gfxterm.c
[INFO ]  =================================================================
[INFO ]  Installing make for build
mv -f term/.deps-core/at_keyboard_module-at_keyboard.Tpo term/.deps-core/at_keyboard_module-at_keyboard.Po
gcc -DHAVE_CONFIG_H -I. -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/grub-core -I..  -Wall -W  -DGRUB_MACHINE_PCBIOS=1 -DGRUB_MACHINE=I386_PC -m32 -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/include -I../include -DGRUB_FILE=\"term/gfxterm_background.c\" -I. -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/grub-core -I.. -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02 -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/include -I../include -I/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/grub-core/lib/libgcrypt-grub/ [... cut ...]
rm: cannot remove '/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/build/.config.2': No such file or directory
make/xtools.make:123: recipe for target '/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/stamp/xtools-build' failed
make: *** [/home/bdx/allcode/github/jamesbeedy/onie_upstream/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/stamp/xtools-build] Error 1
```

full paste here: https://paste.ubuntu.com/p/rXJ5HYGmkB/

Going to try building with the build env container now.

Thank you

jamesbeedy, 2018-05-24:
Building from `build-env` in the container worked flawlessly.

Possibly my host system being ubuntu 18.04, has a different revision of a package then what is expected/used in the debian image and build process.

Not sure if you want to keep this issue open or not for the build failure on the ubuntu host above - I don't care either way.

Thanks again for you help.

closed 2018-05-24 by jamesbeedy

jamesbeedy, 2018-05-24:
for tracking, this build fails similarly on debian and ubuntu bare metal hosts.

abhiranjeet, 2021-12-12:
Hi,

I am facing a similar error while building ONIE for kvm-x86-64. i don't see any error when build is finished. but images directory is empty. any suggestion would be helpful. 

Thanks !