opencomputeproject/onie #757: why i can't install  grub? ERROR: grub-install failed on: /dev/vda

zhuiwang, 2018-06-21:
ONIE:/ # onie-nos-install  http://192.168.1.200:1080/sonic-cavium.bin
discover: installer mode detected.
Stopping: discover... done.
Info: Attempting http://192.168.1.200:1080/sonic-cavium.bin ...
Connecting to 192.168.1.200:1080 (192.168.1.200:1080)
installer            100% |*******************************|   455M  0:00:00 ETA
ONIE: Executing installer: http://192.168.1.200:1080/sonic-cavium.bin
Verifying image checksum ... OK.
Preparing image archive ... OK.
Installing SONiC in ONIE
ONIE Installer: platform: x86_64-cavium-r0
onie_platform: x86_64-kvm_x86_64-r0
deleting partition 3 ...
Filesystem           1K-blocks      Used Available Use% Mounted on
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot.
The operation has completed successfully.
Partition #1 is in use.
Partition #2 is in use.
Partition #3 is available
Creating new SONiC-OS partition /dev/vda3 ...
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot.
The operation has completed successfully.
mke2fs 1.42.13 (17-May-2015)
Creating filesystem with 8388608 4k blocks and 2097152 inodes
Filesystem UUID: be588b7f-76d8-430e-924d-b9e04e821baf
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done   

Installing SONiC to /tmp/tmp.5PocPP/image-master.0-dirty-20180621.003059
Archive:  fs.zip
   creating: boot/
  inflating: boot/config-3.16.0-5-amd64
  inflating: boot/System.map-3.16.0-5-amd64
  inflating: boot/initrd.img-3.16.0-5-amd64
  inflating: boot/config-3.16.0-4-amd64
  inflating: boot/vmlinuz-3.16.0-5-amd64
  inflating: boot/vmlinuz-3.16.0-4-amd64
  inflating: boot/System.map-3.16.0-4-amd64
  inflating: boot/initrd.img-3.16.0-4-amd64
   creating: platform/
  inflating: platform/firsttime
   creating: platform/x86_64-grub/
  inflating: platform/x86_64-grub/grub-pc-bin_2.02~beta2-22+deb8u1_amd64.deb
  inflating: fs.squashfs
ONIE: Unable to find 'Part Number' TLV in EEPROM data.
Success: Support tarball created: /tmp/onie-support-kvm_x86_64.tar.bz2
ERROR: grub-install failed on: /dev/vda
grub-install: error: /usr/lib/grub/x86_64-efi/modinfo.sh doesn't exist. Please specify --target or --directory.
Failure: Unable to install image: http://192.168.1.200:1080/sonic-cavium.bin
ONIE:/ #

cbrune, 2018-06-21:
Hello @zhuiwang -

The relevant error message is here:
```
ERROR: grub-install failed on: /dev/vda
grub-install: error: /usr/lib/grub/x86_64-efi/modinfo.sh doesn't exist. Please specify --target or --directory.
```

There is something wrong with the SONIC installer.

What command line options is the SONIC installer using when invoking `grub-install` ?

zhuiwang, 2018-08-23:
i check the  `/usr/lib/grub` file  ,where don't have  file `x86_64-efi` ,just a `i386-pc`.
however i have succeed install SONiC in an older complie onie-recovery-kvm.bin,which have the file`x86_64-efi`

finaly i found  this in kvm `machin.make`
```
# Enable building of secure boot binaries
SECURE_BOOT_ENABLE = yes
```

```
ifeq ($(SECURE_BOOT_ENABLE),yes)
  GRUB_INSTALL_SB_STAMP	= $(STAMPDIR)/grub-install-sb
  GRUB_MONOLITH_IMAGE	= $(MBUILDDIR)/grub$(EFI_ARCH).efi.unsigned
  GRUB_SECURE_BOOT_IMAGE= $(MBUILDDIR)/grub$(EFI_ARCH).efi
  GRUB_STAMP		+= $(GRUB_INSTALL_SB_STAMP)
  UPDATER_IMAGE_PARTS   += $(GRUB_SECURE_BOOT_IMAGE)
  UPDATER_IMAGE_PARTS_COMPLETE += $(GRUB_INSTALL_SB_STAMP)
  PHONY			+= grub-install-sb
  UEFI_BOOT_LOADER	= $(SHIM_IMAGE_NAME)
else
  UEFI_BOOT_LOADER	= $(GRUB_IMAGE_NAME)
endif

```

```
grub-install-sb: $(GRUB_INSTALL_SB_STAMP)
$(GRUB_INSTALL_SB_STAMP): $(SBSIGNTOOL_INSTALL_STAMP) $(GRUB_INSTALL_STAMP) $(GRUB_HOST_INSTALL_STAMP)
	$(Q) echo "====  Building grub-$(ARCH)-efi-$(GRUB_VERSION) monolithic secure boot image ===="
	$(Q) rm -rf $(SYSROOTDIR)/usr/lib/grub/$(ARCH)-efi
	$(Q) $(SCRIPTDIR)/mk-grub-efi-image $(ARCH) $(GRUB_HOST_BIN_UEFI_DIR) \
		$(GRUB_TARGET_LIB_UEFI_DIR) $(GRUB_MONOLITH_IMAGE)
	$(Q) sbsign --key $(ONIE_VENDOR_SECRET_KEY_PEM) \
		--cert $(ONIE_VENDOR_CERT_PEM) \
		--output $(GRUB_SECURE_BOOT_IMAGE) $(GRUB_MONOLITH_IMAGE)

```
it has remov the "$(SYSROOTDIR)/usr/lib/grub/$(ARCH)-efi" so i cloes the secure boot
```
# Enable building of secure boot binaries
SECURE_BOOT_ENABLE = no
```
after thar , i found the file x86_64-efi, and successfully installed SONiC in ONIE.