opencomputeproject/onie #779: Compile error on DellEMC S4100

evwarner, 2018-10-11:
Get compile error on the Dell EMC S4100 build with the latest August and May versions.  Doesn't really complete compilation until you drop back to the February versions (2018.02br).  The latest compile appears as follows.  What is missing?

gms@debian:~/temp2/onie/build-config$ make -j4 MACHINEROOT=../machine/dell MACHINE=dellemc_s4100_c2338 all
mkdir: created directory '/home/gms/temp2/onie/build/images'
mkdir: created directory '/home/gms/temp2/onie/build/download'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/stamp'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/initramfs'
==== Getting Linux ====
==== Preparing xtools for x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22 ====
==== Getting upstream uClibc ====
==== Getting upstream crosstool-NG ====
==== Preparing a new sysroot directory ====
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/dev'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/sys'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/proc'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/tmp'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/etc'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/lib'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/mnt'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/var'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/var/log'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/usr'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/usr/lib'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/usr/bin'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/usr/sbin'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/usr/share'
mkdir: created directory '/home/gms/temp2/onie/build/dellemc_s4100_c2338-r0/sysroot/usr/share/locale'
mkdir: created directory '/home/gms/temp2/onie/build/user'
mkdir: created directory '/home/gms/temp2/onie/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22'
mkdir: created directory '/home/gms/temp2/onie/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/stamp'
==== Getting upstream zlib ====
2018-10-11 06:32:11 URL:http://mirror.opencompute.org/onie/zlib-1.2.8.tar.gz [571091/571091] -> "/home/gms/temp2/onie/build/download/zlib-1.2.8.tar.gz" [1]
zlib-1.2.8.tar.gz: OK
==== Getting upstream BusyBox ====
2018-10-11 06:32:12 URL:http://mirror.opencompute.org/onie/uClibc-ng-1.0.22.tar.xz [1996808/1996808] -> "/home/gms/temp2/onie/build/download/uClibc-ng-1.0.22.tar.xz" [1]
uClibc-ng-1.0.22.tar.xz: OK
==== Getting upstream util-linux ====
2018-10-11 06:32:12 URL:http://mirror.opencompute.org/onie/crosstool-ng-1.23.0.tar.gz [2237648/2237648] -> "/home/gms/temp2/onie/build/download/crosstool-ng-1.23.0.tar.gz" [1]
crosstool-ng-1.23.0.tar.gz: OK
==== Getting upstream lzo ====
2018-10-11 06:32:12 URL:http://mirror.opencompute.org/onie/busybox-1.25.1.tar.bz2 [2103070/2103070] -> "/home/gms/temp2/onie/build/download/busybox-1.25.1.tar.bz2" [1]
busybox-1.25.1.tar.bz2: OK
==== Getting upstream mtdutils ====
2018-10-11 06:32:12 URL:http://mirror.opencompute.org/onie/e4c8885bddac201ba0ef88560d6444f39e1ff870.tar.gz [345204/345204] -> "/home/gms/temp2/onie/build/download/e4c8885bddac201ba0ef88560d6444f39e1ff870.tar.gz" [1]
e4c8885bddac201ba0ef88560d6444f39e1ff870.tar.gz: OK
==== Getting upstream dropbear ====
2018-10-11 06:32:12 URL:http://mirror.opencompute.org/onie/lzo-2.09.tar.gz [594855/594855] -> "/home/gms/temp2/onie/build/download/lzo-2.09.tar.gz" [1]
lzo-2.09.tar.gz: OK
==== Getting upstream lvm2 ====
2018-10-11 06:32:13 URL:http://mirror.opencompute.org/onie/util-linux-2.27.tar.xz [3968872/3968872] -> "/home/gms/temp2/onie/build/download/util-linux-2.27.tar.xz" [1]
util-linux-2.27.tar.xz: OK
==== Getting upstream dmidecode ====
2018-10-11 06:32:13 URL:http://mirror.opencompute.org/onie/lvm2-2_02_105.tar.xz [999124/999124] -> "/home/gms/temp2/onie/build/download/lvm2-2_02_105.tar.xz" [1]
lvm2-2_02_105.tar.xz: OK
==== Getting upstream ethtool ====
2018-10-11 06:32:13 URL:http://mirror.opencompute.org/onie/dmidecode-3.1.tar.xz [61204/61204] -> "/home/gms/temp2/onie/build/download/dmidecode-3.1.tar.xz" [1]
dmidecode-3.1.tar.xz: OK
==== Getting upstream e2fsprogs ====
2018-10-11 06:32:13 URL:http://mirror.opencompute.org/onie/ethtool-3.14.tar.xz [194360/194360] -> "/home/gms/temp2/onie/build/download/ethtool-3.14.tar.xz" [1]
ethtool-3.14.tar.xz: OK
==== Getting upstream popt ====
2018-10-11 06:32:13 URL:http://mirror.opencompute.org/onie/dropbear-2016.74.tar.bz2 [1622234/1622234] -> "/home/gms/temp2/onie/build/download/dropbear-2016.74.tar.bz2" [1]
dropbear-2016.74.tar.bz2: OK
==== Getting upstream gptfdisk ====
2018-10-11 06:32:13 URL:http://mirror.opencompute.org/onie/popt-1.16.tar.gz [702769/702769] -> "/home/gms/temp2/onie/build/download/popt-1.16.tar.gz" [1]
popt-1.16.tar.gz: OK
==== Getting upstream parted ====
2018-10-11 06:32:13 URL:http://mirror.opencompute.org/onie/gptfdisk-0.8.8.tar.gz [188395/188395] -> "/home/gms/temp2/onie/build/download/gptfdisk-0.8.8.tar.gz" [1]
[... 145 lines cut ...]
Importing unicode...
Importing libgcrypt...
Generating Automake input...
Saving timestamps...
Running autoreconf...
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: AM_GNU_GETTEXT is used, but not AM_GNU_GETTEXT_VERSION
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf
autoreconf: running: /usr/bin/autoheader
autoreconf: running: automake --add-missing --copy --no-force
configure.ac:309: installing 'build-aux/compile'
configure.ac:42: installing 'build-aux/config.guess'
configure.ac:42: installing 'build-aux/config.sub'
configure.ac:48: installing 'build-aux/install-sh'
configure.ac:48: installing 'build-aux/missing'
Makefile.am: installing 'build-aux/depcomp'
parallel-tests: installing 'build-aux/test-driver'
docs/Makefile.am:4: installing 'build-aux/mdate-sh'
docs/Makefile.am:4: installing 'build-aux/texinfo.tex'
autoreconf: Leaving directory `.'
gms@debian:~/temp2/onie/build-config$

evwarner, 2018-10-11:
No, actually figured this one out.  Unlike the Install instructions, perform the 'make' without the multiple jobs parameter, '-j4', and it compiles to completion.

lakesh-sharma, 2018-10-11:
-j12 works fine in our build infra, with 2018.08 based onie.
Good to know that -j4 works fine for you.

evwarner, 2018-10-11:
Actually, -j4 didn’t work at all.  Only when it was omitted entirely would it go through and compile.  Otherwise, it would drop out.

--ericw

cbrune, 2018-10-17:
I am with @lakesh-sharma here -- using `-j<whatever>` should work just fine.  I do it all the time.  Something else is going wrong.

evwarner, 2018-10-24:
Was finally able to drop back into this project.  Also noted the busybox issue, guessing that the busybox patch was for an earlier version of busybox, not updated since the February version where it last worked.
Concerning the make issue, was curious what environment is  working.  Currently using Debian 9.5 (Stretch), with packages brought in to support both ONIE and OpenSwitch.

paulmenzel, 2021-04-02:
Can you please update the original comment and enclose the paste into ```…```, so it’s marked up correctly, and not striked through?

How did you fix the BusyBox patch? Please submit a merge/pull request.

I’d open a new issues for the parallel make issue.

TobiPeterG, 2023-11-25:
I had the same issue. The patch wasn't created, I could "fix" it by editing the busybox.make file and removing one of the top lines defining where the patch file should be, but I couldn't test the image yet, so I don't know if it'll boot.
An actual fix would be nice :)
Could BusyBox be downgraded or the patch be updated?