opencomputeproject/onie #785: Wrong SMBus drivers enabled.

evwarner, 2018-11-01:
Still fighting to get ONIE to compile from the sources of this site to run on a DELL S4128T.  One thing noted in the compiled ONIE is that the TLV header information is not read and attempted installs of OpenSwitch fails.  A running of i2cdetect shows the reason why.  There is no driver for the SMBus controller.  In looking at the kernel patch, the platform-dellemc-s4100-c2338.patch calls 'select I2C_ISCH'.  The issue is that this is the i2c controller for the Intel Menlow, not the Intel Rangeley.  The Rangeley is covered in I2C_I801 (0x8086, 0x1F3C).
It get worse.
Once the ONIE is modified to bring in the correct SMBus driver, the resulting /dev/i2c devices are reversed from what is expected in kernels 4.9.80 and 4.9.95.  The i2c_I801 shows up first at bus 0 instead of bus 1 as the busybox expects, ismt appearing at bus 1.  Believe this to be the case also for the more recent DELL S5100 version as well.

Should be noted that the issue with the busybox patch in dellemc_s4100_c2338 still has not been addressed (Compile error on DellEMC S4100).  Still need to drop back to the February version to get anything to compile and run.

evwarner, 2018-11-01:
Was finally able to get OpenSwitch to load.  The fallacy is that I believed the SMBus driver was needed, which, in a way was true.  But actually, the ismt driver is being used to interrogate the TLVinfo data.  In the original Dell server here running a 4.1.28-based ONIE, there are two i2c devices that come up:  i2c-0 associated with igb_hwmon, indicating the i350 with the system port, and ismt on i2c-1.  In the compiled ONIE versions out at github, only the ismt driver comes up on i2c-0, so the TLV information cannot be read as the compiled busybox is looking for this information on i2c-1.  Now, with the included SMBus driver i2c_I801, at least something is pushing the ismt driver back on i2c-1 where the busybox is expecting it.

lakesh-sharma, 2018-11-05:
I2c ISMT kernel config is enabled by default in platform config.
System eeprom is at 0x53 and is accessible using bus 1 and seems access is just fine
in kernel 4.1.28 and 4.9.95 using i2c_ismt driver.

ONIE:/ # i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- 31 32 -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- 4d -- -- 
50: -- -- -- 53 -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
ONIE:/ # i2cdetect -l
i2c-0   i2c             igb BB                                  I2C adapter
i2c-1   smbus           SMBus iSMT adapter at dff14000          SMBus adapter
ONIE:/ # uname -a
Linux onie 4.1.28-onie+ #1 SMP Thu Aug 9 11:38:08 PDT 2018 x86_64 GNU/Linux
ONIE:/ #

evwarner, 2018-11-05:
This is correct.  Below you see that i2c-0 is taken up by the igb HWMON driver in 4.1.28, which is not turned on in the ONIE out at github with kernels 4.9.80 and 4.9.95.  This causes the i2c_ismt driver to come up at i2c-0 instead of i2c-1 as busybox is expecting, which therefore cannot read the TLV info.

--ericw