opencomputeproject/onie #831: xtools failed

giltabibian, 2019-05-01:
Hi,

I'm trying to build ONIE for new platform machine (not included current machine).
running on Ubuntu 14.04 with latest onie branch: origin/2019.02.01br

before the build I update all necessary packages:
sudo apt-get update
make debian-prepare-build-host

make -j4 MACHINEROOT=../machine/<vendor> MACHINE=<vendor>_<model> distclean
make -j4 MACHINEROOT=../machine/<vendor> MACHINE=<vendor>_<model> all

this is the log, failed on xtools build. any ideas? 

====  Building xtools for armv7a-g6.3.0-lnx4.4.30-uClibc-ng-1.0.22 ====
gmake[1]: Entering directory `/local/store/projects/onie-github/build/x-tools/armv7a-g6.3.0-lnx4.4.30-uClibc-ng-1.0.22/build'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20190501.105702
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 1.45s (at 00:05)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 130.35s (at 02:16)
[INFO ]  =================================================================
[INFO ]  Installing make for build
[INFO ]  Installing make for build: done in 35.50s (at 02:51)
[INFO ]  =================================================================
[INFO ]  Installing m4 for build
[ERROR]    /local/store/projects/onie-github/build/x-tools/armv7a-g6.3.0-lnx4.4.30-uClibc-ng-1.0.22/build/build/src/m4-1.4.18/lib/strerror-override.h:53:38: error: expected identifier or '(' before 'void'
[ERROR]    /local/store/projects/onie-github/build/x-tools/armv7a-g6.3.0-lnx4.4.30-uClibc-ng-1.0.22/build/build/src/m4-1.4.18/lib/strerror-override.h:53:38: error: expected ')' before numeric constant
[ERROR]    make[5]: *** [Makefile:1910: strerror-override.o] Error 1
[ERROR]    make[4]: *** [Makefile:1674: all] Error 2
[ERROR]    make[3]: *** [Makefile:1572: all-recursive] Error 1
[ERROR]    make[2]: *** [Makefile:1528: all] Error 2
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing m4 for build'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@338]
[ERROR]  >>        called from: do_m4_backend[scripts/build/companion_tools/100-m4.sh@66]
[ERROR]  >>        called from: do_companion_tools_m4_for_build[scripts/build/companion_tools/100-m4.sh@17]
[ERROR]  >>        called from: do_companion_tools_for_build[scripts/build/companion_tools.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@653]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/local/store/projects/onie-github/build/crosstool-ng/crosstool-ng-1.23.0/docs/B - Known issues.txt'
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]  
[ERROR]  (elapsed: 4:02.91)
gmake[1]: *** [build] Error 2
gmake[1]: Leaving directory `/local/store/projects/onie-github/build/x-tools/armv7a-g6.3.0-lnx4.4.30-uClibc-ng-1.0.22/build'
rm: cannot remove ?/local/store/projects/onie-github/build/x-tools/armv7a-g6.3.0-lnx4.4.30-uClibc-ng-1.0.22/build/.config.2?: No such file or directory
make: *** [/local/store/projects/onie-github/build/x-tools/armv7a-g6.3.0-lnx4.4.30-uClibc-ng-1.0.22/stamp/xtools-build] Error 1

giltabibian, 2019-05-01:
this issue occurs also with docker solution suggested before:
https://github.com/opencomputeproject/onie/tree/master/contrib/build-env

by the way, is there any good reason for onie to compile standalone toolchain? 
can't I compile onie project with pre compiled toolchain?

build log with docker:
[INFO ]  Installing binutils for host
[ERROR]    make[7]: *** [Makefile:720: install-binPROGRAMS] Error 1
[ERROR]    make[6]: *** [Makefile:1180: install-am] Error 2
[ERROR]    make[5]: *** [Makefile:1004: install-recursive] Error 1
[ERROR]    make[4]: *** [Makefile:1174: install] Error 2
[ERROR]    make[3]: *** [Makefile:3677: install-binutils] Error 2
[ERROR]    make[2]: *** [Makefile:2256: install] Error 2
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing binutils for host'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@338]
[ERROR]  >>        called from: do_binutils_backend[scripts/build/binutils/binutils.sh@229]
[ERROR]  >>        called from: do_binutils_for_host[scripts/build/binutils/binutils.sh@97]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@653]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/home/build/src/onie/build/crosstool-ng/crosstool-ng-1.23.0/docs/B - Known issues.txt'
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]  
[ERROR]  (elapsed: 10:41.85)
/home/build/src/onie/build/crosstool-ng/crosstool-ng-1.23.0/ct-ng:147: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/home/build/src/onie/build/x-tools/armv8a-g6.3.0-lnx4.4.30-glibc-2.25/build'
rm: cannot remove '/home/build/src/onie/build/x-tools/armv8a-g6.3.0-lnx4.4.30-glibc-2.25/build/.config.2': No such file or directory
make/xtools.make:123: recipe for target '/home/build/src/onie/build/x-tools/armv8a-g6.3.0-lnx4.4.30-glibc-2.25/stamp/xtools-build' failed
make: *** [/home/build/src/onie/build/x-tools/armv8a-g6.3.0-lnx4.4.30-glibc-2.25/stamp/xtools-build] Error 1
build@0bb6d53d2f7f:~/src/onie/build-config$

neuralsandwich, 2023-03-01:
I'm also seeing this error

mshych, 2023-03-01:
What's your build branch?
Did you see the notes: https://opencomputeproject.github.io/onie/developers/building.html#branches-and-build-environment-compatibility ?
Do you build in docker /DUE? If yes, which version?
You can also try to make a direct build on the Debian machine: https://opencomputeproject.github.io/onie/developers/building.html#preparing-a-new-build-machine

jon-nokia, 2023-09-21:
I also ran into this issue with the docker build and solved it by installing rsync in the docker image.  Look in x-tools...build.log for "rsync: not found"