opencomputeproject/onie #848: "make -j4 MACHINE=kvm_x86_64 all recovery-iso" does not build the recovery image.

sweatypolarbears, 2019-06-08:
Hi, I have been trying to build the recovery image for onie x86-64 kvm. But the binaries are not being created in the "build/images" directory and the folder is empty. While looking at the log, I noticed that a fatal error is occuring and that several packages are missing. The error is as follows:

make -f /home/somak/onie/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/pesign/pesign-0.112/Make.deps deps SOURCES="cms_common.c content_info.c oid.c password.c signed_data.c signer_info.c ucs2.c  pesign_context.c actions.c client.c efikeygen.c efisiglist.c siglist.c pesigcheck.c pesigcheck_context.c certdb.c pesign.c pesign_context.c actions.c daemon.c"
Package efivar was not found in the pkg-config search path.
Perhaps you should add the directory containing `efivar.pc'
to the PKG_CONFIG_PATH environment variable
No package 'efivar' found
Package nss was not found in the pkg-config search path.
Perhaps you should add the directory containing `nss.pc'
to the PKG_CONFIG_PATH environment variable
No package 'nss' found
Package nspr was not found in the pkg-config search path.
Perhaps you should add the directory containing `nspr.pc'
to the PKG_CONFIG_PATH environment variable
No package 'nspr' found
Package popt was not found in the pkg-config search path.
Perhaps you should add the directory containing `popt.pc'
to the PKG_CONFIG_PATH environment variable
No package 'popt' found
make[3]: Entering directory '/home/somak/onie/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/pesign/pesign-0.112/src'
echo SOURCES=cms_common.c content_info.c oid.c password.c signed_data.c signer_info.c ucs2.c  pesign_context.c actions.c client.c efikeygen.c efisiglist.c siglist.c pesigcheck.c pesigcheck_context.c certdb.c pesign.c pesign_context.c actions.c daemon.c
SOURCES=cms_common.c content_info.c oid.c password.c signed_data.c signer_info.c ucs2.c pesign_context.c actions.c client.c efikeygen.c efisiglist.c siglist.c pesigcheck.c pesigcheck_context.c certdb.c pesign.c pesign_context.c actions.c daemon.c
make[3]: Leaving directory '/home/somak/onie/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/pesign/pesign-0.112/src'
/usr/bin/gcc -O0 -g3 -fvar-tracking -fvar-tracking-assignments -Wall -Werror -Wextra  -Wall -Werror -Wno-cpp  -Wsign-compare -Wno-unused-result -Wno-unused-function -std=gnu11 -fshort-wchar -fPIC -flto -fno-strict-aliasing -fno-merge-constants -fkeep-inline-functions -D_GNU_SOURCE -DCONFIG_x86_64 -I/home/somak/onie/build/user/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/pesign/pesign-0.112/include      -Werror  -c -o authvar.o authvar.c
Package efivar was not found in the pkg-config search path.
Perhaps you should add the directory containing `efivar.pc'
to the PKG_CONFIG_PATH environment variable
No package 'efivar' found
Package nss was not found in the pkg-config search path.
Perhaps you should add the directory containing `nss.pc'
to the PKG_CONFIG_PATH environment variable
No package 'nss' found
Package nspr was not found in the pkg-config search path.
Perhaps you should add the directory containing `nspr.pc'
to the PKG_CONFIG_PATH environment variable
No package 'nspr' found
Package popt was not found in the pkg-config search path.
Perhaps you should add the directory containing `popt.pc'
to the PKG_CONFIG_PATH environment variable
No package 'popt' found
authvar.c:22:18: fatal error: popt.h: No such file or directory
compilation terminated.

I am trying this on an Ubuntu 16.04 LTS desktop. Any help would be appreciated.

ehdoyle, 2019-06-13:
Hi - it looks like you've got a missing package problem. I was able to build in Ubuntu 16.04 (well, a Docker container of it, but that _should_ be the same thing) after installing the build dependencies:
If you check onie/build-config/Makefile, you should find this:
DEBIAN_BUILD_HOST_PACKAGES  = build-essential stgit u-boot-tools util-linux \ 
                  gperf device-tree-compiler python-all-dev xorriso \         
                  autoconf automake bison flex texinfo libtool libtool-bin \  
                  realpath gawk libncurses5 libncurses5-dev bc \              
                  dosfstools mtools pkg-config git wget help2man libexpat1 \  
                  libexpat1-dev fakeroot python-sphinx rst2pdf \              
                  libefivar-dev libnss3-tools libnss3-dev libpopt-dev \       
                  libssl-dev sbsigntool uuid-runtime uuid-dev cpio \          
                  bsdmainutils                                                

Try typing the following:
cd onie/build-config
make distclean
sudo apt-get install < list of packages starting with build-essential and ending with bsdmainutils> 
make -j4 MACHINE=kvm_x86_64 all recovery-iso

...all that should clean out whatever had been built, install the build dependencies, and build clean.
Let me know how it goes,
-Alex

sweatypolarbears, 2019-06-17:
Hi I followed your instructions and it worked perfectly. You were right about the mssing packages. Thank you!!!

closed 2019-06-17 by sweatypolarbears