opencomputeproject/onie #853: ONIE build problem on Debian10 (Buster)

mshych, 2019-07-23:
There is a problem to build ONIE on build server with new Debian 10 release.
1.  The first problem comes in config stage during a check of bash version.
configure.ac should be changed in order to check also versions 5 and up.
It was checked only against versions 3 .x and 4. 
Debian10 has bash version 5.0.3

Fix:
['^GNU bash, version (3\.[1-9]|4|5)'])

2. After that, there is a failure in building xtools.
"
====  Building xtools for x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22 ====
make[1]: Entering directory '/hdd/build/michaelsh/onie/mlnx_5.2.00_onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/build'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20190723.105140
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 0.54s (at 00:02)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 100.59s (at 01:43)
[INFO ]  =================================================================
[INFO ]  Installing make for build
[INFO ]  Installing make for build: done in 20.83s (at 02:04)
[INFO ]  =================================================================
[INFO ]  Installing m4 for build
[ERROR]    /hdd/build/michaelsh/onie/mlnx_5.2.00_onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/build/build/src/m4-1.4.18/lib/freadahead.c:92:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
[ERROR]    make[5]: *** [Makefile:1910: freadahead.o] Error 1
[ERROR]    make[4]: *** [Makefile:1674: all] Error 2
[ERROR]    make[3]: *** [Makefile:1572: all-recursive] Error 1
[ERROR]    make[2]: *** [Makefile:1528: all] Error 2
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing m4 for build'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@338]
[ERROR]  >>        called from: do_m4_backend[scripts/build/companion_tools/100-m4.sh@66]
[ERROR]  >>        called from: do_companion_tools_m4_for_build[scripts/build/companion_tools/100-m4.sh@17]
[ERROR]  >>        called from: do_companion_tools_for_build[scripts/build/companion_tools.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@653]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/hdd/build/michaelsh/onie/mlnx_5.2.00_onie/build/crosstool-ng/crosstool-ng-1.23.0/docs/B - Known issues.txt'
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
"

Regards,
    Michael.

ehdoyle, 2019-07-26:
Hi Michael - what target were you trying to build? Or more specifically, what was the 'make' command used to start the build? I tried a build with a Buster container for mlnx_x86 and the virtual machine, and they didn't exhibit the problem. 
Thanks,
-Alex

mshych, 2019-07-28:
Hi Alex,

I’m building mlnx_x86 target.
However, I doubt that this is related to specific machine build as problems appeared a long time before specific machine build, during getting packages, config and building crosstools.
Please note that I don’t build in docker but directly on Debian build server with Buster release.

Regards,
    Michael.

ehdoyle, 2019-07-31:
Hi Michael, 
 I finally had a chance to test this on a system with Buster installed (vs the container) and hit the bash check issue, which looks like it's solved in the latest version of crosstool-ng
https://github.com/crosstool-ng/crosstool-ng/blob/master/configure.ac
...but it looks like updating this touches a few things, so in terms of complexity it is a bit more than a version bump. 
So the short is I'd definitely stick with building in Stretch for now, I'll see if I can figure out why my Buster container build worked, and this bug should stay open as I'm expecting I'll hit a few more bugs doing a Buster tools update.

dracuka, 2020-04-29:
wangdongwen@debian:~/tmp/onie-newest/onie/build/x-tools/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.22/build/build/src/m4-1.4.18$ sed -i 's/IO_ftrylockfile/IO_EOF_SEEN/' lib/*.c
wangdongwen@debian:~/tmp/onie-newest/onie/build/x-tools/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.22/build/build/src/m4-1.4.18$ 
wangdongwen@debian:~/tmp/onie-newest/onie/build/x-tools/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.22/build/build/src/m4-1.4.18$ 
wangdongwen@debian:~/tmp/onie-newest/onie/build/x-tools/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.22/build/build/src/m4-1.4.18$ 
wangdongwen@debian:~/tmp/onie-newest/onie/build/x-tools/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.22/build/build/src/m4-1.4.18$ echo "#define _IO_IN_BACKUP 0x100" >> lib/stdio-impl.h

kielfriedt, 2021-02-25:
I am building on x86 running `$ make MACHINE=kvm_x86_64 all` and hit this same errors on Ubuntu 18 & 20. 
 Bash issue is corrected by @mshych  fix.

```
[INFO ]  Installing m4 for build
[ERROR]    /home/ubuntu/onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/build/build/src/m4-1.4.18/lib/freadahead.c:92:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
[ERROR]    make[5]: *** [Makefile:1910: freadahead.o] Error 1
[ERROR]    make[4]: *** [Makefile:1674: all] Error 2
[ERROR]    make[3]: *** [Makefile:1572: all-recursive] Error 1
[ERROR]    make[2]: *** [Makefile:1528: all] Error 2
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing m4 for build'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@338]
[ERROR]  >>        called from: do_m4_backend[scripts/build/companion_tools/100-m4.sh@66]
[ERROR]  >>        called from: do_companion_tools_m4_for_build[scripts/build/companion_tools/100-m4.sh@17]
[ERROR]  >>        called from: do_companion_tools_for_build[scripts/build/companion_tools.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@653]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/home/ubuntu/onie/build/crosstool-ng/crosstool-ng-1.23.0/docs/B - Known issues.txt'
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]
[ERROR]  (elapsed: 1:40.04)
make[1]: *** [/home/ubuntu/onie/build/crosstool-ng/crosstool-ng-1.23.0/ct-ng:147: build] Error 2
make[1]: Leaving directory '/home/ubuntu/onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/build'
rm: cannot remove '/home/ubuntu/onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/build/.config.2': No such file or directory
make: *** [make/xtools.make:125: /home/ubuntu/onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/stamp/xtools-build] Error 1
```

paulmenzel, 2021-04-01:
@ehdoyle, I guess your container image, the package *bash* was not installed.

Anyway, with *bash* installed, I am able to reproduce this on Ubuntu 20.10.

ksbomj, 2021-06-19:
Hi guys,

Have the same problem, build for accton_as4610_54 which is arm machine. Any way to solve it? Or maybe someone can suggest a distro which is safe to build this?

ehdoyle, 2021-12-31:
I believe this has been resolved in the master branch, which does build on Debian 10 (or at least in Debian 10 containers configured by DUE - individual setups on real hardware will, obviously, vary). Machine targets that haven't upgraded to the master branch can still build using Debian 9 DUE containers off the 2021.08br branch.
I'll close this out in a few days unless there's anything else to add. Thanks!

ehdoyle, 2022-01-11:
Closing out.

closed 2022-01-11 by ehdoyle

zhengbiqing, 2022-03-20:
> I am building on x86 running `$ make MACHINE=kvm_x86_64 all` and hit this same errors on Ubuntu 18 & 20. Bash issue is corrected by @mshych fix.
>
> ```
> [INFO ]  Installing m4 for build
> [ERROR] [...]

kielfriedt, do you fix the problem: 
rm: cannot remove '/home/ubuntu/onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.22/build/.config.2': No such file or directory