opencomputeproject/onie #859: Can't install Open Network Linux on ONIE kvm x86-64

sweatypolarbears, 2019-08-19:
Hi!! I have installed the ONIE kvm x86-64 on my desktop by following the build instructions provided here. I can log into the ONIE CLI fine. However I have been encountering several errors related to the grub installation when I'm trying to install ONL on it. The log is as follows:

ONIE:/ # onie-nos-install http]://opennetlinux.org/binaries/latest-DEB8-AMD64-IN
STALLED.installer
discover: installer mode detected.
Stopping: discover... done.
Failure: Unknown URL type: http]://opennetlinux.org/binaries/latest-DEB8-AMD64-INSTALLED.installer
ONIE:/ # onie-nos-install http://opennetlinux.org/binaries/latest-DEB8-AMD64-INS
TALLED.installer
discover: installer mode detected.
Stopping: discover... done.
Info: Attempting http://opennetlinux.org/binaries/latest-DEB8-AMD64-INSTALLED.installer ...
Connecting to opennetlinux.org (165.227.87.248:80)
installer            100% |*******************************|   214M  0:00:00 ETA
ONIE: Executing installer: http://opennetlinux.org/binaries/latest-DEB8-AMD64-INSTALLED.installer
installer: computing checksum of original archive
installer: checksum is OK
installer: extracting pad
1+0 records in
1+0 records out
512 bytes (512B) copied, 0.000021 seconds, 23.3MB/s
installer: copying file before resetting pad
installer: resetting pad
1+0 records in
1+0 records out
512 bytes (512B) copied, 0.000020 seconds, 24.4MB/s
installer: extracting shar into /tmp/sfx-fHnBl7
installer: invoking installer installer.sh
Found installer tmpfs on /tmp/sfx-fHnBl7 (/tmp) using opts rw,noatime
Resizing tmpfs /tmp to 1048576k
Unpacking ONL installer files...
Extracting from /tmp/sfx-fHnBl7/onie-installer.zip: onl-loader-initrd-amd64.cpio.gz ...
Archive:  /tmp/sfx-fHnBl7/onie-installer.zip
  inflating: onl-loader-initrd-amd64.cpio.gz
Extracting initrd to /tmp/sfx-fHnBl7/initrd-7L3K4k
65674 blocks
Setting up /dev
Setting up /run
Setting up mounts
modprobe: can't change directory to '/lib/modules': No such file or directory
Launching ONL installer
Unmounting /mnt/onie-boot (--force)
Unmounting /boot/efi (--force)
Extracting from /tmp/sfx-fHnBl7/onie-installer.zip: preinstall.sh ...
Archive:  /tmp/sfx-fHnBl7/onie-installer.zip
  inflating: preinstall.sh
Invoking pre-install actions
Hello from preinstall
Chroot is /tmp/sfx-fHnBl7/initrd-7L3K4k
Examining /dev/vda6 --> ONL-DATA
Examining /dev/vda5 --> ONL-IMAGES
Examining /dev/vda4 --> ONL-CONFIG
Examining /dev/vda3 --> ONL-BOOT
Examining /dev/vda2 --> ONIE-BOOT
Found ONIE boot partition at /dev/vda2
Examining /dev/vda1 --> EFI System
Found EFI System partition at /dev/vda1
Found a clean GPT partition table
getting installer configuration
+ blkid
found ONIE boot device /dev/vda2
found ONIE boot mounted at /mnt/onie-boot
found ONIE initrd at /mnt/onie-boot/onie/initrd.img-4.9.95-onie
extracting initrd /mnt/onie-boot/onie/initrd.img-4.9.95-onie
+ /bin/mkdir /tmp/sfx-fHnBl7/chroot-i_Dkj6.d
+ xz -dc /mnt/onie-boot/onie/initrd.img-4.9.95-onie | cpio -imd
63140 blocks
preparing chroot in /tmp/sfx-fHnBl7/chroot-i_Dkj6.d
+ mount -v -t proc proc /tmp/sfx-fHnBl7/chroot-i_Dkj6.d/proc
+ mount -v -t sysfs sysfs /tmp/sfx-fHnBl7/chroot-i_Dkj6.d/sys
[... 382 lines cut ...]
installer failed
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/onl/install/App.py", line 212, in runLocalOrChroot
    code = self.installer.run()
  File "/usr/lib/python2.7/dist-packages/onl/install/BaseInstall.py", line 845, in run
    return self.installGpt()
  File "/usr/lib/python2.7/dist-packages/onl/install/BaseInstall.py", line 825, in installGpt
    code = self.installGrub()
  File "/usr/lib/python2.7/dist-packages/onl/install/BaseInstall.py", line 750, in installGrub
    self.im.grubEnv.install(self.partedDevice.path)
  File "/usr/lib/python2.7/dist-packages/onl/install/ConfUtils.py", line 233, in install
    _install()
  File "/usr/lib/python2.7/dist-packages/onl/install/ConfUtils.py", line 226, in _install
    + [device,])
  File "/usr/lib/python2.7/dist-packages/onl/install/InstallUtils.py", line 1254, in check_call
    subprocess.check_call(cmd, *args, cwd=cwd, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 511, in check_call
    raise CalledProcessError(retcode, cmd)
CalledProcessError: Command '['chroot', '/tmp/sfx-fHnBl7/chroot-ts0Ju6.d', 'grub-install', '--boot-directory=/tmp/mount-IbQENA.d', '--target=x86_64-efi', '--no-nvram', '--recheck', '--bootloader-id=ONL', '--efi-directory=/tmp/mount-Ry4HK0.d', u'/dev/vda']' returned non-zero exit status 1
re-attaching to tty
entering Python debugger (installer_debug=1)
> /usr/lib/python2.7/subprocess.py(511)check_call()
-> raise CalledProcessError(retcode, cmd)

I opened the usr/lib/grub directory and found no folder named "x86_64-efi", only the "id386-pc" folder. Is there any way to remedy this? I am running the kvm on Ubuntu 16.04 LTS

dingrui37, 2019-11-12:
The same problem @sweatypolarbears 
```
installer failed
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/onl/install/App.py", line 212, in runLocalOrChroot
    code = self.installer.run()
  File "/usr/lib/python2.7/dist-packages/onl/install/BaseInstall.py", line 845, in run
    return self.installGpt()
  File "/usr/lib/python2.7/dist-packages/onl/install/BaseInstall.py", line 825, in installGpt
    code = self.installGrub()
  File "/usr/lib/python2.7/dist-packages/onl/install/BaseInstall.py", line 750, in installGrub
    self.im.grubEnv.install(self.partedDevice.path)
  File "/usr/lib/python2.7/dist-packages/onl/install/ConfUtils.py", line 233, in install
    _install()
  File "/usr/lib/python2.7/dist-packages/onl/install/ConfUtils.py", line 226, in _install
    + [device,])
  File "/usr/lib/python2.7/dist-packages/onl/install/InstallUtils.py", line 1254, in check_call
    subprocess.check_call(cmd, *args, cwd=cwd, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 511, in check_call
    raise CalledProcessError(retcode, cmd)
CalledProcessError: Command '['chroot', '/tmp/sfx-XqNGEq/chroot-dVGSk0.d', 'grub-install', '--boot-directory=/tmp/mount-SzYrdP.d', '--target=x86_64-efi', '--no-nvram', '--recheck', '--bootloader-id=ONL', '--efi-directory=/tmp/mount-SmGNQD.d', u'/dev/vda']' returned non-zero exit status 1
re-attaching to tty
entering Python debugger (installer_debug=1)
> /usr/lib/python2.7/subprocess.py(511)check_call()
-> raise CalledProcessError(retcode, cmd)

```

liuhycn, 2021-03-11:
@sweatypolarbears 
Hi, I encounter the same problem. 

May I ask that how do you config your VM to access the Internet? 

Thank you!