opencomputeproject/onie #881: update make to use ct-ng-1.24.0 to fix bug 676934

adaraiseh, 2019-12-04:
Trying to build ONIE kvm version on Debian 10.2, it keeps on breaking with error:
checking for GNU bash >= 3.1... no
configure: error: Required tool not found: GNU bash >= 3.1

Looking for this problem seems crosstool-NG has bug that got fixed on 1.24.0 version:
676934 sys-devel/ct-ng-1.23.0 : configure: error: Required tool not found: GNU bash >= 3.1

Is there a way to update it manually ? tried to change crosstool-ng.make file manually to download 1.24.0 instead of 1.23.0 but that also failed.

ehdoyle, 2020-01-06:
Hi - there's currently a number of components that need to be updated to work on Buster - and crosstool-ng is going to be a bit tricky as ONIE has a couple of patches for it ( vs just bumping a revision, which is what most people tend to think of when updating a component. )
.
However you can work around this by using Docker.
I've got a DUE [https://github.com/ehdoyle/DUE/tree/master/templates/onie](https://github.com/ehdoyle/DUE/tree/master/templates/onie) template that creates an ONIE build environment in a Docker container, and it's how I built the last quarterly release, so it's had some testing. (though it's still a 1.0 release, so there are probably rough edges)

If you're already on a Debian system that has packaging tools you can:
## Install DUE
1. ` git clone https://github.com/ehdoyle/DUE`
2.  `cd DUE`
3.  `dpkg-buildpackage -uc -us`
4.  `sudo dpkg -i ../due_1.0.0_all.deb`
...and then maybe add yourself to the Docker group if it wasn't already installed.

## Build  the container
`./due --create --from debian:9 --description "ONIE Build" --name onie-build --prompt ONIE --tag onie --use-template onie`
(More info on the container specifics here: )
[https://github.com/ehdoyle/DUE/tree/master/templates/onie](https://github.com/ehdoyle/DUE/tree/master/templates/onie)

## Use the build environment

1.  Have the ONIE source checked out in your home directory
2.  due --run  (and, there being only 1 container, it won't ask you to choose )
3.  You're now in the container, and it has mounted your home directory
4.  cd onie/build-config
5.  build.  ( can test with `make -j4 MACHINE=kvm_x86_64 all` )
If you've got a system with Docker installed, you can create an ONIE build container using these tools ( which is what I used for the last quarterly release) 

It's a 1.0 release, so there are probably some rough edges.   Just looking at the ReadMe.md it could use some cleanup, but should be enough to get you going.
There's also a [https://github.com/ehdoyle/DUE/blob/master/docs/Troubleshooting.md](https://github.com/ehdoyle/DUE/blob/master/docs/Troubleshooting.md) guide for errors you might encounter.
If you end up trying this, let me know how it works out and if there are any areas that aren't clear/need updating so I can improve it.
Thanks!
Alex Doyle
ONIE Project Lead

ehdoyle, 2021-12-31:
Update: ct-ng has been upgraded to 1.24.0 in the master branch, and the 2021.08br branch should build systems that are using the older 1.23 version using a Debian 9 build environment. This process is described in the documentation
[here](https://opencomputeproject.github.io/onie/developers/building.html#preparing-an-onie-build-environment)
Unless there's anything else to add I'll close this out in a few days. Thanks!

ehdoyle, 2022-01-11:
Closing out.

closed 2022-01-11 by ehdoyle