opencomputeproject/onie #909: Erroro compiling quanta_common_p2020 image

plant75, 2020-04-19:
Compilig on a debian 9 i have this error:

[INFO ]  Installing pass-1 core C gcc compiler
[ERROR]    make[3]: *** [Makefile:2107: s-attrtab] Killed
[ERROR]    make[3]: *** Waiting for unfinished jobs....
[ERROR]    make[2]: *** [Makefile:3955: all-gcc] Error 2
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-1 core C gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@338]
[ERROR]  >>        called from: do_gcc_core_backend[scripts/build/cc/100-gcc.sh@674]
[ERROR]  >>        called from: do_gcc_core_pass_1[scripts/build/cc/100-gcc.sh@227]
[ERROR]  >>        called from: do_cc_core_pass_1[scripts/build/cc.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@653]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      '/home/onie/onie/build/crosstool-ng/crosstool-ng-1.23.0/docs/B - Known issues.txt'
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]  
[ERROR]  (elapsed: 5:34.46)
/home/onie/onie/build/crosstool-ng/crosstool-ng-1.23.0/ct-ng:147: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/home/onie/onie/build/x-tools/powerpc-softfloat-g4.9.2-lnx3.2.69-uClibc-ng-1.0.22/build'
rm: cannot remove '/home/onie/onie/build/x-tools/powerpc-softfloat-g4.9.2-lnx3.2.69-uClibc-ng-1.0.22/build/.config.2': No such file or directory
make/xtools.make:123: recipe for target '/home/onie/onie/build/x-tools/powerpc-softfloat-g4.9.2-lnx3.2.69-uClibc-ng-1.0.22/stamp/xtools-build' failed
make: *** [/home/onie/onie/build/x-tools/powerpc-softfloat-g4.9.2-lnx3.2.69-uClibc-ng-1.0.22/stamp/xtools-build] Error 1

[build.log](https://github.com/opencomputeproject/onie/files/4498734/build.log)

ehdoyle, 2020-04-20:
Hi - I was able to get this to build with:
make distclean ; make -j4 MACHINEROOT=../machine/quanta MACHINE=quanta_common_p2020 all demo
If you're still seeing the failure, try setting up a Docker container that's preconfigured to build ONIE, which is the build environment I use for development and release.
The code and instructions are here:
https://github.com/CumulusNetworks/DUE/tree/master/templates/onie

ehdoyle, 2020-05-14:
I'll close this out for now. Feel free to reopen or reach out if you are still having problems with this.

closed 2020-05-14 by ehdoyle