opencomputeproject/onie #910: Compiling Cisco IOS XR ONIE Installer Image

bashmaz, 2020-04-28:
Hi,

I am trying to compile ONIE Compatible installer for IOS XR to be installed on Edgecore AS5916-54XKS. 

I am not able to find a clear document for that purpose.

I would appreciate your kind support.

Thanks.

ehdoyle, 2020-04-30:
Hi,
 Sorry to say that unless the code is in the repository, there's not much to be done - there aren't any Edgecore switches in the platforms that have been upstreamed, so that's not even available for reference.
For ONIE specific questions, you might try the mailing list: OCP-ONIE@OCP-All.groups.io
Good luck!

rkandilarov, 2020-05-05:
@bashmaz , I also came to similar problem thinking that the brand is `Edgecore` have a look at `Accton` switches ([Edgecore Networks Corporation is a wholly-owned subsidiary of Accton Technology Corporation](https://www.garlandtechnology.com/edgecore-networks)). What you want is most probably `accton_as5916_54xks` from `onie/machine/accton/accton_as5916_54xks`.

Hope building with `MACHINE=accton_as5916_54xks` will do the trick for you!

ehdoyle, 2021-12-30:
I'll close this out. I just confirmed that the accton_as5916_54xks does build off the 2021.08br legacy branch, in a Debian 9 DUE container environment, and noted that Edgecore is a subsidary of Accton.

commit 68df2a90 (2021-12-30): onie-build-targets.json - confirmed build of Accton as5916_54xks

Build environment: A Debian 9 DUE container and the 2021.08br branch
Build command: make -j4 MACHINEROOT=../machine/accton \
                 MACHINE=accton_as5916_54xks all demo

As Edgecore is owned by Accton, this build may work for Edgecore
switches with a matching model number.

Closes #910
closed 2021-12-30 by ehdoyle