opencomputeproject/onie #912: Update fsck/e2fsck to at least 1.43 or 1.44.5 to support metadata_csum

rkandilarov, 2020-05-05:
Hello!

Today I got issue with ext4 disk partition on device running Cumulus 4.0. As Cumulus 4 is based on  Debian Buster (version 10) when installed it used the Ext4 Metadata Checksum capability. When checking the disk using onie-rescue:
```
ONIE:/ #  fsck /dev/sda5
fsck 1.42.13 (17-May-2015)
e2fsck 1.42.13 (17-May-2015)
/dev/sda5 has unsupported feature(s): metadata_csum
e2fsck: Get a newer version of e2fsck!
```
I did updated to the latest ONIE, compile (it take me some time to get around the DUE) just to realize that ONIE is using e2fsprogs-1.42.13.tar.xz (dummy me, not to look at the source code directly).

According to https://packages.debian.org/buster/e2fsprogs the used version in Buster, so in Cumulus 4.* is 1.44.5-1 .

And finally to my question - can I change the used by onie version in `build-config/make/e2fsprogs.make` to `E2FSPROGS_VERSION=1.44.5` and force downloading 
https://mirrors.edge.kernel.org/pub/linux/kernel/people/tytso/e2fsprogs/v1.44.5/e2fsprogs_1.44.5.orig.tar.gz  or it is not that simple?

Regards!

rkandilarov, 2020-05-06:
Changing to e2fsck 1.44.5-1 was not so straight forward. There need to be applied also one patch to btrfs  (https://patchwork.kernel.org/patch/10281327/ ) and to disable native language support (`--disable-nls`) in the Makefile `build-config/make/e2fsprogs.make`. At least this helps ONIE to compile. After update on the switch I got its file system checked - a short demo follows: 
```
ONIE:/ # fsck.ext4 -f -c -v /dev/sda5
e2fsck 1.44.5 (15-Dec-2018)
sh: badblocks: not found
CL-SYSTEM: Updating bad block inode.
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information

CL-SYSTEM: ***** FILE SYSTEM WAS MODIFIED *****

       60911 inodes used (13.41%, out of 454176)
          74 non-contiguous files (0.1%)
          36 non-contiguous directories (0.1%)
             # of inodes with ind/dind/tind blocks: 0/0/0
             Extent depth histogram: 56989/49
      566390 blocks used (30.35%, out of 1865979)
           0 bad blocks
           1 large file

       50706 regular files
        6223 directories
          55 character device files
          25 block device files
           0 fifos
           0 links
        3892 symbolic links (3784 fast symbolic links)
           1 socket
------------
       60902 files
ONIE:/ # 
```

In case anybody is interested, I could submit a pull request, but... I have not gone deep into the changes between the  e2fsck versions, so a overview from "kernel" person is needed.

P.S. There seems to be a file `onie/patches/btrfs-progs/0001-configure-refactor-backtrace-detection.patch` that is not listed in the corresponding `series`. Probably a leftover that needs deletion as it cannot be applied.

ehdoyle, 2020-05-14:
Hi - I'd absolutely be interested in looking at a pull request or ( a set of diffs from where you've done the development work) . Component version upgrade and building with newer Linux releases ( Debian 10  in particular) are going to be areas I'm focusing on now that I've managed to set up a Jenkins instance to rebuild everything and track the impact of changes.)

rkandilarov, 2020-05-14:
@ehdoyle - here you are: https://github.com/opencomputeproject/onie/pull/915 .
Regards!

ehdoyle, 2020-05-15:
Wow that was faster than I expected. It may take me a bit to get to looking at this as my job has some build tooling that needs to be deployed in a hurry but I will come back to this. Thanks!

ehdoyle, 2021-12-31:
Hi - with the upgrades in the master branch, e2fsprogs has been upgraded to version 1.46.3, which hopefully should address this. I'll close this out in a few days unless there are additional issues. Thanks!

ehdoyle, 2022-01-11:
Closing out.

closed 2022-01-11 by ehdoyle