opencomputeproject/onie #937: Errors building Onie for both Celestica and Edge-core

chenowethNTX, 2020-12-22:
I have a paid of Edge-core 7512-32x 100gbe switches and a Celestica d4040.
Getting different errors depending on what build I'm doing.
For the edge core, the simple error is:
Makefile:208: /machine.make: No such file or directory
Makefile:213: Unsupported switch ASIC vendor: unknown
Makefile:214: Check the setting of SWITCH_ASIC_VENDOR in machine.make
Makefile:215: *** Supported switch ASIC vendors: bcm bft cavium centec mlnx mvl nephos none qemu.  Stop.
I've cleaned up with make distclean, and still getting errors.
Any help here would be greatly appreciate. Losing my mind trying to get this to work.

Thank you
[log_cel.txt](https://github.com/opencomputeproject/onie/files/5727854/log_cel.txt)
[log_ec.txt](https://github.com/opencomputeproject/onie/files/5727855/log_ec.txt)

ehdoyle, 2020-12-24:
Hi - looking at the logs you don't have stgit installed, which provides the missing /usr/bin/stg.
If you have a Debian  system running Docker, or one where it can be installed, I'd suggest using Dedicated User Environment to create an ONIE build container that has all of the build dependencies built in. It's what I've been using for everyday development and the quarterly releases.
https://github.com/CumulusNetworks/DUE/tree/master/templates/onie
It's pretty much:
1. git clone https://github.com/CumulusNetworks/DUE.git
2. cd DUE
3. make depends  (will install Docker.io on your host system if you don't already have it)
3a. Add yourself to the Docker group, log out and back in for it to take effect.
4. ./due --create --from debian:9 --description "ONIE Build Debian 9" --name onie-build-debian-9 --prompt ONIE-9 --tag onie --use-template onie
5. due --run
6. ...and then you're in a Debian 9 ONIE build environment.

chenowethNTX, 2020-12-29:
> Hi - looking at the logs you don't have stgit installed, which provides the missing /usr/bin/stg.
> If you have a Debian system running Docker, or one where it can be installed, I'd suggest using [...]

Thank you, any ideas on the edge-core 7512 errors?

ehdoyle, 2020-12-29:
Hi - I'd say it's probably the same problem as they're failing to find stg during package download :)
(Which is not to say that there might be other problems, but fixing things here may solve them.)

chenowethNTX, 2020-12-30:
I built a new debian environment per your recommendation and tried building one natively and with due and I'm struggling.
I'm able to get the due environment up (due --run) but I have no idea what to do after that.
building it normally without using DUE, the celestica and even testing out just kvm VM seems to work, but they aren't completing an ISO? with the recovery-iso.
I'm just completely lost here.

ehdoyle, 2020-12-30:
Sounds like there may be a number of problems here, around running DUE, build environment setup, and build invocation. 
Q - what version of Linux are you running DUE on?
Q  - if DUE was installed via the .deb or a 'make install', you should be able to just run it with 'due' below. Otherwise you'll have to run it out of the directory it was cloned in to with ./due
 A  - I think you're typing 7512-32, rather than 7512_32
 
This should work, provided your home directory isn't nfs mounted (if it is, there's a slightly more complicated workaround, as Docker gets weird mounting network mounted fileysystems)
cd ~/
git clone https://github.com/CumulusNetworks/DUE.git
cd DUE
./due --create --from debian:9 --description "ONIE Build Debian 9" --name onie-build-debian-9 --prompt ONIE-9 --tag onie --use-template onie
## now you've got the Debian 9 build environment for ONIE
cd ..
git clone https://github.com/opencomputeproject/onie.git
## latest ONIE present as a clean checkout
cd ../DUE
./due --run-image due-onie-build-debian-9    <- prompt should now be due-onie-build-debian-9 now that you're in the container and an 'ls' should list your home directory. 
cd onie/build-config
make MACHINEROOT=../machine/accton/ MACHINE=accton_as7512_32x all recovery-iso demo
...and that should work.
If you don't want to build in the container, you can invoke the build with:
cd onie/build-config
due --run-image  due-onie-build-debian-9 --command make MACHINEROOT=../machine/accton/ MACHINE=accton_as7512_32x all recovery-iso demo

I just tried that locally, so I believe it should work - let me know how it goes.

chenowethNTX, 2020-12-30:
I've tried this on an old build of federa 20, believe it's been updated. On a new build of Fedora 33 I think. After your suggestion, I built a debian build on LTS and installed DUE there.
I'm running into a group/user problem with docker on the debian build that docker isn't a group inside the container? Does that make sense? I'll see if I can pull up the message. I've added the group and my user to the group, but having issues once I've ran due --run and/or ./due --create --from debian:9 --description "ONIE Build Debian 9" --name onie-build-debian-9 --prompt ONIE-9 --tag onie --use-template onie and inside the one build environment.
For the Edge-core 7512's with Due or without this is the message I get for building them:
make -j4 MACHINEROOT=../machine/accton/ MACHINE=accton_as7512-32x all recovery-iso
bash: due: command not found
Makefile:213: Unsupported switch ASIC vendor: unknown
Makefile:214: Check the setting of SWITCH_ASIC_VENDOR in machine.make
Makefile:215: *** Supported switch ASIC vendors: bcm bft cavium centec mlnx mvl nephos none qemu. Stop.
with DUE for the Celestica D4040 i am getting this error:
checking for bash... /usr/bin/bash
checking for GNU bash >= 3.1... no
configure: error: Required tool not found: GNU bash >= 3.1
make: *** [make/crosstool-ng.make:86: /home/ronald/onie/build/crosstool-ng/stamp/crosstool-ng-configure] Error 1

I cleaned the build directory and on the fedora 20 box, the accton finally attempted building, however I got this error message:
checking for makeinfo... no
configure: error: missing required tool: makeinfo
make: *** [/root/onie/build/crosstool-ng/stamp/crosstool-ng-configure] Error 1
which is new.
so that is two different systems, two different errors. The new debian box doesn't recognize the accton machine data, and errors on bash version. The old fedora box gets makeinfo error

ehdoyle, 2020-12-31:
DUE on Red Hat hasn't been tested much beyond me getting to work on a local system, so there is probably some room for improvement there - still, after:
 checking DUE out
 running a 'make install', 
 building a Debian 9 ONIE container
 ...and logging in to that container with due --run I'm able to cd to the onie/build-config directory and build with
 
I think there's a couple of places where you may be going off track.
-  If you're getting 'unsupported asic' you've probably mistyped the machine name - double check the '-' vs '_' in the as7512_32 in your make command.
- Can you double check that you're a member of the Docker group on the system? I've frequently found I have to log out and back in again after adding myself to a group for the changes to take effect. 
 That said, there are some striking implementation differences in programs between Debian and Red Hat, particularly around privilege escalation (needed to be root in the container) which is why DUE only creates Debian images doesn't create images based off of Red Hat...yet
 - Can you provide more context around the command issued and error messages returned?
 'bash: due: command not found' could mean that you executed in the container ( where due is not installed ) or on the host system, and due was never 'make install' ed for a Red Hat system
 -Any messages around failing to find a program indicate you're _definitely_ not executing in the context of the container.
Thanks.

chenowethNTX, 2021-01-01:
so a weird thing. i've added myself to the group for docker and logged out, however when I'm in the DUE docker environment I get the message that I'm not part of that group, if I try to add it, I get the message that the group docker doest exist.
I've attempted to create it in the docker shell to no luck.

chenowethNTX, 2021-01-01:
for this: "If you're getting 'unsupported asic' you've probably mistyped the machine name - double check the '-' vs '_' in the as7512_32 in your make command." I'm literally copying and pasting from here or github or a txt file I have saved with he commands.
for this: "Can you provide more context around the command issued and error messages returned?"
when running the make command or the due make command, on the debian system I get that bash error.
"root@debian:/home/ronald/onie/build-config# make MACHINEROOT=../machine/celestic                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             a/ MACHINE=cel_smallstone_xp all recovery-iso demo
==== Configuring crosstool-NG ====
Running autoconf...
Done. You may now run:
    ./configure
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking for ggrep... no
checking for grep... grep
checking for absolute path to grep... /usr/bin/grep
checking for gegrep... no
checking for egrep... egrep
checking for absolute path to egrep... /usr/bin/egrep
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for sed... /usr/bin/sed
checking for GNU sed >= 4.0... yes
checking whether ln -s works... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking how to run the C preprocessor... gcc -E
checking if gcc can static link... yes
checking for ranlib... ranlib
checking for gobjcopy... no
checking for objcopy... objcopy
checking for absolute path to objcopy... /usr/bin/objcopy
checking for gobjdump... no
checking for objdump... objdump
checking for absolute path to objdump... /usr/bin/objdump
checking for greadelf... no
checking for readelf... readelf
checking for absolute path to readelf... /usr/bin/readelf
checking for gperf... gperf
checking for absolute path to gperf... /usr/bin/gperf
checking for bison... bison
checking for flex... flex
checking for makeinfo... makeinfo
checking for cut... cut
checking for readlink... readlink
checking for tar... tar
checking for gzip... gzip
checking for bzip2... bzip2
checking for xz... xz
checking for help2man... help2man
checking for wget... wget
checking for curl... no
checking for stat... stat
checking whether stat takes GNU or BSD format... GNU
checking whether to use getconf or sysctl to count CPUs... getconf _NPROCESSORS_                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             ONLN
checking for gpatch... no
checking for patch... patch
checking for absolute path to patch... /usr/bin/patch
checking for bash... /usr/bin/bash
checking for GNU bash >= 3.1... no
configure: error: Required tool not found: GNU bash >= 3.1
make: *** [make/crosstool-ng.make:86: /home/ronald/onie/build/crosstool-ng/stamp                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             /crosstool-ng-configure] Error 1
root@debian:/home/ronald/onie/build-config#"

ehdoyle, 2021-01-01:
So for those last two - you won't see due or Docker in the container - those are the applications used to run a container that is configured to be a virtual Debian 9 system with everything installed to build ONIE - and the make should work at that point - though I'd suggest a clean checkout for the code you're trying as you shouldn't have to build the code as root and that may introduce some issues, though it looks like things haven't got there yet.
The bash issue you're hitting may be the one described here, as, outside of the container your host system doesn't have the required component versions:
https://github.com/opencomputeproject/onie/issues/881

chenowethNTX, 2021-01-02:
root@debian:/home/ronald/DUE# dpkg-buildpackage -uc -us
dpkg-buildpackage: error: cannot open file debian/changelog: No such file or directory
root@debian:/home/ronald/DUE# ^C
root@debian:/home/ronald/DUE# sudo dpkg -i ../due_1.0.0_all.deb
dpkg: error: cannot access archive '../due_1.0.0_all.deb': No such file or directory
checking that

chenowethNTX, 2021-01-04:
Alex, i've been going through my installs and rechecking everything, you were absolutely correct that I was using -32x instead of _32x, I've also made sure to double-check all the required pieces to build/make.
I'm seeing this currently and never seen it before, so I'm hoping this is what I should be expecting??
ronald@localhost:~$ make -j4 MACHINEROOT=../machine/accton/ MACHINE=accton_as751                                                                                                                                                                  2_32x all recovery-iso demo
make: *** No rule to make target `all'.  Stop.
ronald@localhost:~$
ronald@localhost:~$ cd onie
ronald@localhost:~/onie$ cd build-config/
ronald@localhost:~/onie/build-config$ make -j4 MACHINEROOT=../machine/accton/ MA                                                                                                                                                                  CHINE=accton_as7512_32x all recovery-iso demo
====  Building xtools for x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.22 ====
====  Patching grub-2.02 ====
error: could not lock config file /home/ronald/onie/build/user/x86_64-g4.9.2-lnx                                                                                                                                                                  4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/.git/config: Permission denied
error: could not lock config file /home/ronald/onie/build/user/x86_64-g4.9.2-lnx                                                                                                                                                                  4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/.git/config: Permission denied
error: could not lock config file /home/ronald/onie/build/user/x86_64-g4.9.2-lnx                                                                                                                                                                  4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/.git/config: Permission denied
Reinitialized existing Git repository in /home/ronald/onie/build/user/x86_64-g4.                                                                                                                                                                  9.2-lnx4.9.95-uClibc-ng-1.0.22/grub/grub-2.02/.git/
fatal: Unable to create '/home/ronald/onie/build/user/x86_64-g4.9.2-lnx4.9.95-uC                                                                                                                                                                  libc-ng-1.0.22/grub/grub-2.02/.git/index.lock': Permission denied
make: *** [/home/ronald/onie/build/user/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.22                                                                                                                                                                  /stamp/grub-patch] Error 128
make: *** Waiting for unfinished jobs....
make[1]: Entering directory `/home/ronald/onie/build/x-tools/x86_64-g4.9.2-lnx4.                                                                                                                                                                  9.95-uClibc-ng-1.0.22/build'
make[1]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule                                                                                                                                                                  .
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20210104.071951
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 0.66s (at 00:                                                                                                                                                                  03)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components

[INFO ]  Extracting and patching toolchain components: done in 281.69s (at 04:44)
[INFO ]  =================================================================
[INFO ]  Installing make for build
[INFO ]  Installing make for build: done in 32.62s (at 05:17)
[INFO ]  =================================================================
[INFO ]  Installing m4 for build
[INFO ]  Installing m4 for build: done in 66.46s (at 06:23)
[INFO ]  =================================================================
[INFO ]  Installing autoconf for build
[INFO ]  Installing autoconf for build: done in 7.28s (at 06:31)
[INFO ]  =================================================================
[INFO ]  Installing automake for build
[INFO ]  Installing automake for build: done in 5.78s (at 06:36)
[INFO ]  =================================================================
[INFO ]  Installing libtool for build
[INFO ]  Installing libtool for build: done in 23.85s (at 07:00)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[INFO ]  Installing ncurses for build: done in 36.40s (at 07:37)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 72.66s (at 08:49)
[INFO ]  =================================================================
[... cut ...]
[INFO ]  =================================================================
[INFO ]  Installing gettext for host
[INFO ]  Installing gettext for host: done in 0.01s (at 11:36)
[INFO ]  =================================================================
[INFO ]  Installing binutils for host
[INFO ]  Installing binutils for host: done in 135.42s (at 13:51)
[INFO ]  =================================================================
[INFO ]  Installing pass-1 core C gcc compiler
obviously it's not done yet, but I'm hopeful!?
Thank you

ehdoyle, 2021-01-07:
Hi - just to follow up - did you get it to work?
Also - were there points in the documentation that could have been clearer? I'm trying to combat 'author bias' where it is totally clear to me what I'm talking about, but not so clear to someone encountering it for the first time.

chenowethNTX, 2021-01-09:
I haven't got it working completely yet.
I feel that for a novice it's pretty straight forward.

chenowethNTX, 2021-01-09:
``ronald@debian:~/onie/build-config$ due --run-image due-onie-build-debian-9 --command make MACHINEROOT=../machine/accton/ MACHINE=accton_as7512_32x all recovery-iso demo
-
 ----------------------------------------------------------------------------
 ----------------------------------------------------------------------------
 ------ : start and log in to [ 34f0af536c56 ] with this command:      --
   docker run --rm\
     --volume /home/ronald:/home/ronald\
     --volume /home/ronald/onie/build-config:/home/ronald/onie/build-config\
     --name ronald_due-onie-build-debian-9_1610174151\
     --hostname due-onie-build-debian-9 34f0af536c56 /usr/local/bin/container-create-user.sh\
     --docker-group-id 125\
     --username ronald\
     --userid 1000\
     --groupid 1000\
     --groupname ronald\
     --command cd /home/ronald/onie/build-config ; make MACHINEROOT=../machine/accton/ MACHINE=accton_as7512_32x all recovery-iso demo

 ----------------------------------------------------------------------------
 ------ Host file systems are mounted as follows: ---------------------------
 -
 -   --volume /home/ronald:/home/ronald
 -   --volume /home/ronald/onie/build-config:/home/ronald/onie/build-config
 -
 ----------------------------------------------------------------------------
 -------- Type 'exit' to leave.  Docker Container Output Follows ------------

 ___________________________________________________________________________
|                                                                           |
| ronald : Creating user UID [ 1000 ] GID [ 1000 ] Group [ ronald ]
| ronald : /home directory exists. Not creating.
| ronald : adding to sudoers file.
|___________________________________________________________________________|
|                                                                           |
|               Appending container /etc/due-bashrc                         |
|___________________________________________________________________________|
|___________________________________________________________________________
|
| Running [ cd /home/ronald/onie/build-config ; make MACHINEROOT=../machine/accton/ MACHINE=accton_as7512_32x all recovery-iso demo ]
|  ...as user [ ronald ] UID [ 1000 ]
|__________________________________________________________________________
bash: make: command not found
 ___________________________________________________________________________
|                                                                           |
| Done   [ cd /home/ronald/onie/build-config ; make MACHINEROOT=../machine/accton/ MACHINE=accton_as7512_32x all recovery-iso demo ]
| Status [ 127 ]
|___________________________________________________________________________|
ronald@debian:~/onie/build-config$"

ehdoyle, 2021-01-11:
Looks like the build command is correct. I tried it and hit an error at the very end making the iso because it couldn't find the mkdosfs library.
Try logging in to the container and building:
due --run-image due-onie-build-debian-9
cd /home/ronald/build-config
make MACHINEROOT=../machine/accton/ MACHINE=accton_as7512_32x all recovery-iso demo

The failure to find make would imply that there's something wrong with the container build, or the path in the container, so I'm puzzled.
I'd think any container problems would have prevented it from building in the first place.

chenowethNTX, 2021-01-15:
I'm still no further along with this.
Noticed/realized two of the systems I've using root. I went and added users and working my way through giving them sudo privileges and figuring out the permissions for the build commands I installed as root.
Still no further than before sadly.

ehdoyle, 2021-01-16:
From the limited information it's tough to figure out what steps you might be missing. The only other thing I can think of is to start with a fresh system and walk through the instructions for setting up a DUE ONIE build container - that is going to be less complicated than the alternatives.
There's a tutorial video here that goes from source download to KVM build, and that might illustrate where the problem is.
[https://www.youtube.com/watch?v=-5onRbZA0QQ](https://www.youtube.com/watch?v=-5onRbZA0QQ)

chenowethNTX, 2021-01-19:
will do

chenowethNTX, 2021-02-03:
looks like I was finally successful in getting this made last night for KVM:

chenowethNTX, 2021-02-03:
kvm_x86_64-r0.initrd   kvm_x86_64-r0.vmlinuz.unsigned          onie-updater-x86_64-kvm_x86_64-r0
kvm_x86_64-r0.vmlinuz  onie-recovery-x86_64-kvm_x86_64-r0.iso
ronald@due-onie-build-debian-9:~/onie/build/images$

ehdoyle, 2021-02-03:
Excellent! If there were any spots in the documentation that could have been clearer (or are subject to multiple interpretations) let me know and I'll try and make it clearer.

chenowethNTX, 2021-02-03:
once I was able to sit down with your youtube video and step by step it, it was fantastic and straight forward.
lord knows how many webpages i've had open with details on building the images.
so one last question, obviously I can put the r0.iso file on a tfp/web server and let onie connect to it for discovery/imaging.
If I use a USB stick, i assume i'm putting:
kvm_x86_64-r0.initrd
kvm_x86_64-r0.vmlinuz.unsigned
onie-updater-x86_64-kvm_x86_64-r0
kvm_x86_64-r0.vmlinuz
in the root and do I put the iso in a certain directory or root?

Thank you

ehdoyle, 2021-02-04:
Glad to hear it. 
You should be able to burn the iso to usb and boot off of it.

chenowethNTX, 2021-02-05:
Perfect, thank you so much Alex!

chenowethNTX, 2021-02-06:
Alex, is the demo different than a full build?
I assume if so, you'd just do all for the iso?

chenowethNTX, 2021-02-07:
I've installed the Demo, and this may be out of your wheel house and may need another ticket opened, but I don't apparently have onie working on this switch currently.
Is this up your alley?
ACCTON_AS7512_32X-OS:/ # onie-start
/bin/onie-start: line 8: /etc/init.d/discover.sh: not found

ehdoyle, 2021-02-09:
Hi - the per-platform stuff I'm generally not familiar with, as that lands in the realm of the manufacturer andI probably can't get access to the hardware anyway. NOS specific stuff is definitely more the realm of the NOS vendor, as ONIE has some suggestions about getting invoked, and thats it.  If an ONIE build supports a demo-os target, then that binary should be installable on the hardware from a running ONIE, using onie-nos-install <demo*bin>

chenowethNTX, 2021-02-11:
Thank you Alex, so one last question, what is the update/recovery files for then? are they just to load/reload onie or are they spefically for the switch being built for? just trying to understand what the significance of the three files types are.
Thank you

ehdoyle, 2021-02-12:
Sure - update revolves around upgrading an existing onie, and the recovery is for initial install, or reinstall.
Running the kvm build illustrates this, where it uses QEMU to boot off the recovery iso image and it installs ONIE in to a blank qcow2 image that becomes the 'switch filesystem'
There is documentation here:
https://github.com/opencomputeproject/onie/blob/master/machine/kvm_x86_64/INSTALL
..and I'm working on a kvm automation tool that'll just run things since I'm always missing a qemu argument, but the command line ends up looking something like this:
qemu-system-x86_64                                                                  
 --enable-kvm                                                                       
 -m 1024                                                                          
 -name onie                                                                         
 -vnc 0.0.0.0:244                                                                   
 -device virtio-net,netdev=onienet,mac=52:54:00:13:34:EF                            
 -netdev user,id=onienet,hostfwd=::-:22                                             
 -boot order=cd,once=d                                                              
 -cdrom /home/adoyle/onie/kvm/onie-recovery-x86_64-kvm_x86_64-r0.iso                
 -drive file=/home/adoyle/onie/kvm/onie-x86-demo.qcow2,media=disk,if=virtio,index=0 
 -serial telnet:localhost:9200,server

chenowethNTX, 2021-02-13:
Gotha, thank you Alex

closed 2021-02-13 by chenowethNTX