opencomputeproject/onie #940: ONIE 2020.11 fails to build with `MACHINE=qemu_armv8a`

jmpolom, 2021-01-13:
Attempting to build ONIE 2020.11 in a Debian Stretch build container for the `qemu_armv8a` machine results in build failure. Build fails during kernel build as it fails to find `qemu_armv8a.dtb`:

"""
make[2]: *** No rule to make target 'arch/amd64/boot/dts/qemu_armv8a.dtb'. Stop.
arch/arm64/Makefile:94 recipe for target 'qemu_armv8a.dtb' failed
"""

Those are the most relevant sections of the output, and that's where things get interesting. Taking a look through [onie/machine/qemu_armv8a](https://github.com/opencomputeproject/onie/tree/master/machine/qemu_armv8a) gives the impression this is not actually a supported machine and is incompletely implemented. The kernel config is empty as is the u-boot config, which I suspect should at least both have _something_ in them as they're populated with content for [qemu_armv7a](https://github.com/opencomputeproject/onie/tree/master/machine/qemu_armv7a).

I also perused kernel sources to attempt to locate a device tree file for `qemu_armv8a` and struck out. It seems like there never has been nor is a `qemu_armv8a.dts` in the kernel tree. The device tree isn't included in a patch like it is for [qemu_armv7a](https://github.com/opencomputeproject/onie/blob/master/machine/qemu_armv7a/kernel/platform-qemu_armv7a.patch) again leading me to believe that support for this machine is not really implemented.

Can anyone confirm whether ONIE should in fact build for machine type `qemu_armv8a`?

ehdoyle, 2021-01-13:
I can't recall seeing it build, and this audit of the current state of builds would support that:
[https://github.com/opencomputeproject/onie/blob/34be58fd2f849ee7c61a8b03e58a8cc129d04989/build-config/scripts/onie-build-targets.json#L4](https://github.com/opencomputeproject/onie/blob/34be58fd2f849ee7c61a8b03e58a8cc129d04989/build-config/scripts/onie-build-targets.json#L4)
I'd consider getting that to build to be the second highest priority - right behind getting component upgrades stabilized, as arm64 systems are becoming increasingly common, but any work on that is a ways off...unless of course someone gets ambitious and submits a pull request :)

lguohan, 2021-04-25:
wonder if there is any update on this? @ehdoyle , is this on you highest priority now?

ehdoyle, 2021-04-26:
My highest priority - absolutely. 
My employer's highest priority for me...not so much :)
However I'm glad to see this - I've been wanting to see this build for well over a year.
Code changes are being pushed off between the latest release candidate and the next quarterly release, so it is a moot point for the next few weeks, by which point I'll hopefully be out of Insane Crunch Mode at work...

closed 2021-08-19 by ehdoyle