opencomputeproject/onie #945: Make on Buster fails in Configuring crosstool-NG

alpauna, 2021-03-03:
using:
make -j4 MACHINEROOT=../machine/celestica MACHINE=cel_redstone all
Results in error:
==== Configuring crosstool-NG ====
Running autoconf...
./bootstrap: 5: ./bootstrap: autoconf: not found
make: *** [make/crosstool-ng.make:86: /home/apauna/code/onie-master/build/crosstool-ng/stamp/crosstool-ng-configure] Error 127

looked in stamp dir mentioned above and there is no bootstrap file in that folder. Also notice no crosstool-ng-configure only crosstool-ng-patch and crosstool-ng-source exist. in the crosstool-ng folder there is a crosstool-ng-1.23.0 and that appears to have bootstrap in that dir. 

Is there problems building this on Buster?

I have attached the output from the make as well as txt file

[busterrestonebuilderrors.txt](https://github.com/opencomputeproject/onie/files/6078515/busterrestonebuilderrors.txt)

ehdoyle, 2021-03-05:
Hi - yes, there's a build issue for buster, but you do have a couple of options:

The component-upgrade-2020 branch does build on Buster, but hasn't been widely tested yet, so it is probably not what you want.

ONIE does build for almost all most platforms in a Debian 9 Stretch environment, but since it is a bit much to expect people to have one just lying around, I've been using Docker containers to create Debian 9 (ONIE release and build ) and Debian 10 (component upgrade work) ONIE build environents.
They're set up with a Docker wrapper program called 
[Dedicated User Environment](https://github.com/CumulusNetworks/DUE/tree/master/templates/onie), which sets up the container to be identical to what is used for ONIE releases, and has some convenient features for running builds inside the container (like mounting your home directory and creating an identical user account in the container, so you have a lot of the user configuration from the host system )
You can dpkg -i the latest release .deb  [here](https://github.com/CumulusNetworks/DUE/releases/tag/v2.3.0)
and run a sudo apt-get install --fix-broken to reslove the dependencies, or just check out the source code and run it locally.

paulmenzel, 2021-03-26:
https://github.com/opencomputeproject/onie/issues/853 also mentions problems building on Debian 10 (buster).

paulmenzel, 2021-04-01:
@alpauna, reading the error message ` ./bootstrap: 5: ./bootstrap: autoconf: not found`, you probably need to install the package *autoconf*: `sudo apt install autoconf`.

Please check that, and close the issue, if that solves it. Maybe even update the issue title and add `./bootstrap: autoconf: not found` there.

PS: GitHub issues also support [GitHub Markdown](https://guides.github.com/features/mastering-markdown/) improving legibility greatly.

ehdoyle, 2021-12-31:
Hi - just checking if the issues resolved. If so I'll close this out.

ehdoyle, 2022-01-11:
Closing out.

closed 2022-01-11 by ehdoyle