opencomputeproject/onie #946: dropbear version is very outdated

user318, 2021-03-09:
Hello,

I suggest to update the version of dropbear. The version in ONIE is 2016.74, which is almost 5 years old. Current version 2020.81 have better support for ciphers, etc. For example it has support for ed25515 keys and chacha20-poly1305 cipher.
I have attached some patch, that I made for myself.
[onie-dropbear-update.patch.txt](https://github.com/opencomputeproject/onie/files/6111603/onie-dropbear-update.patch.txt)
Couple of words about what is there.
It works for me, but it needs some additional work, I think. Because I have disabled there some ciphers and dss, but those may be required by others.
Dropbear now have different structure of included config file, so you do not need to copy all the defaults there and it needs to be placed as localoptions.h. The defines also have 0/1 values now instead of being set or not.
I have added "--disable-harden" to configure, because otherwise the build procces was breaking with an error about zlib was missing. But in config.log I saw it because ld complaint about some ssp library. Do not know exactly what it is. I suggested it is related to stack protection and found out the "--disable-harden" option and it worked out.

paulmenzel, 2021-03-26:
If you have a patch, it’d be great if you submitted a merge/pull request.

user318, 2021-04-09:
Ok, I'll do it. Just, as I said, I am not sure my solution is good for general use, so I decided just to show what I mean.

ehdoyle, 2022-01-11:
Closing out this bug as the pull request has been merged and seems to be working well.
[here](https://github.com/opencomputeproject/onie/commit/00a3761fdd891ce8ac90b07abdfa66bdac302dfc)

closed 2022-01-11 by ehdoyle