opencomputeproject/onie #951: `onie-mk-itb.sh: line 182: mkimage: command not found`

paulmenzel, 2021-04-01:
The built U-Boot tool `mkimage` is not added to the path, so the build fails with

    /dev/shm/onie/build-config/scripts/onie-mk-itb.sh: line 182: mkimage: command not found

The tool is built by U-Boot:

    /dev/shm/onie/build/delta_tn48m-r0/u-boot/mvebu_db_armada8k/tools/mkimage

paulmenzel, 2021-04-01:
Same for `/dev/shm/onie/build/delta_tn48m-r0/u-boot/mvebu_db_armada8k/scripts/dtc/dtc`:

    sh: dtc: command not found
    mkimage: Can't open /dev/shm/onie/build/images/delta_tn48m-r0.itb.tmp: No such file or directory
    make: *** [make/images.make:339: /dev/shm/onie/build/images/delta_tn48m-r0.itb] Error 255

ehdoyle, 2021-12-30:
Hi - sorry it's taken 8 months to get back to this.

 I got the image to build using the 2021.08br branch of ONIE using
**make -j4 MACHINEROOT=../machine/delta MACHINE=delta_tn48m all**
in a Debian 9 DUE [ONIE container](https://github.com/CumulusNetworks/DUE/tree/master/templates/onie)
Created with the 3.0 version of DUE using:
**./due --create --from debian:9  --description "ONIE Build Debian 9" --name onie-build-debian-9 --prompt ONIE-9 --tag onie --use-template onie**
I'll add that the DemoOS did NOT build, though, and I'll note that.
Are you still having issues with this?

commit ba80e4b4 (2021-12-30): onie-build-targets.json - update status of delta_tn48m

Noting that the 'all' target builds. The demo target does not.
Confirmed to build in a Debian 9 DUE container with the
2021.08br  branch.

Closes #951
closed 2021-12-30 by ehdoyle