opencomputeproject/onie #954: Missing machine Accton/Edgecore AS5114 (AS4224)

paulmenzel, 2021-04-07:
We got an Accton/Edgecore AS5114, but the ONIE support is not upstream. @david56, do you know more about the upstreaming status?

```
# fw_printenv
arch=arm
autoload=no
baudrate=115200
board=mvebu_armada-8k
board_name=mvebu_armada-8k
boot_diag=if test -n $onie_boot_reason; then if test $onie_boot_reason = diag; then run diag_bootcmd; fi; fi
bootcmd=run boot_diag; run check_boot_reason; run nos_bootcmd; run onie_bootcmd
bootdelay=10
check_boot_reason=if test -n $onie_boot_reason; then setenv onie_bootargs boot_reason=$onie_boot_reason; run onie_bootcmd; fi;
console=console=ttyS0,115200 earlycon=uart8250,mmio32,0xf0512000
consoledev=ttyS0
cpu=armv8
cpuidle=cpuidle.off=1
dhcp_user-class=arm64-accton_as4224-r0_uboot
dhcp_vendor-class-identifier=arm64-accton_as4224-r0
diag_bootcmd=setenv bootargs console=$consoledev,$baudrate $diagargs ; run diag_setup; scsi scan; sleep 4 && ext2load scsi 0:3 $loadaddr $diag_image && bootm $loadaddr
diag_image=accton_as4224_52-00.0b.01.01-diag
diag_setup=setenv initrd_high 0xFFFFFFFFFFFFFFFF; setenv loadaddr 0x20000000
diagargs=console=ttyS0,115200 earlycon=uart8250,mmio32,0xf0512000 pci=pcie_bus_safe mem=2048M
eth1addr=34:EF:B6:EC:36:3C
eth2addr=34:EF:B6:EC:36:3D
ethaddr=34:EF:B6:EC:36:3B
ethprime=eth1
extra_params=pci=pcie_bus_safe
fdt_high=0xffffffffffffffff
fdt_name=fdt.dtb
fdtcontroladdr=7f5ed6a8
gatewayip=10.4.50.254
get_images=tftpboot $kernel_addr_r $image_name; tftpboot $fdt_addr_r $fdt_name; run get_ramfs
get_ramfs=if test "${ramfs_name}" != "-"; then setenv ramdisk_addr_r 0x8000000; tftpboot $ramdisk_addr_r $ramfs_name; else setenv ramdisk_addr_r -;fi
hostname=marvell
image_name=Image
ipaddr=0.0.0.0
loadaddr=0x20000000
netdev=eth1
netmask=255.255.255.0
onie_args=run onie_initargs onie_platformargs
onie_dropbear_dss_host_key=begin-base64@600@d#AAAAB3NzaC1kc3MAAACBALuuOOHQH9K3gIykG7IFlci2R8ABkCC1bWA9x63N#kvRyZdNIgr37s9xuITOSEfBwgcTvr5CWkYV+ocSLo6YHXWWj4pRxPUV4MTP/#Gg8pMKWTChqsInOU5WkRK7TlqLgwdOIQjkH0+De4Zx87JyosEUJP9w4NDIXy#J+NmDZqpDGXzAAAAFQDwizDCTSj2+793zCH3P+nal3SaNwAAAIEAia+rqbAK#giki1o0MuRUFj9pS8509+nDnSgNdeyD90CEzvv41EPzzA6KZ+19flBvBJLPu#4lTwqiaQBpLZ+s2vxaqG1nX0xpDXIDkf8F5QGCEfwl372/5rMDsMMm/tiwgd#ljYdkxEzJfZNEN09bGeW5UfcEw5fu9jqvdIrvmpRvV0AAACALxd6SGqsdUev#ZhB2mV7Zj/eIfaTfzjWjwy29V4V/0Ns+YNTTWjyi1j7egx9oCc9LQ/aHdAHF#qkTGYNGbbFk3bCsaJu2gNfONnjUA/jlKRoKuc0Q5unKqFKyJHdbw+DsxbAhZ#hwKKO3cuPWKgJ9BlD7SzWQRswGm7FVHgesrxRz8AAAAUD13/5neE/m9uDDFi#MN48MAuNWF8=#====#
onie_dropbear_ecdsa_host_key=begin-base64@600@r#AAAAE2VjZHNhLXNoYTItbmlzdHAyNTYAAAAIbmlzdHAyNTYAAABBBK1PLle1#B4e0xsdPOJFsLrZzM/VTzXoeE62ugqTQx262VRw9NNLt1bL1Ws5fgVi6D84E#tEOGPMhfKY37EST5HzMAAAAhALtmhLxMoDOObymgykXYp9B4zGEyZ4aT55M5#5am44VPh#====#
onie_dropbear_rsa_host_key=begin-base64@600@r#AAAAB3NzaC1yc2EAAAADAQABAAAAgQCTAVjsti2C0vdna9uYLoZqhZlsue0E#PY8uIu3vSsW5fx+4Vb69cnRuiym8QXXwmHl1XpFTfND7Y8ui+AbXhuJ0V0Zv#5jU6YbW/gzpiifpVds+LgEYYfgEL8BE/q0Gj0EhngPONASc7ScSue6brh8m1#yMjAt+nFZsSg4ahm+EspvQAAAH9dy3HbH0Zruw8sSZs9yKhKYF6VZCmGixyA#YlQqxx0qgOpvLxI01YAsXvzwClgGs4MRFXdKmVtcyftGxtwJBog8XIxIi3Et#ZlmFJiyxWQ16gfrhEEqW/acDKtEBDZs1RSQ8neQ1IzezH5VmLQHrreEkD0Wa#cJBOV+QTU/ZV1BThAAAAQQC4Yl9XweV7XjrA4CCfyUX0p0w9fLmdrmC4whBW#39/UiSiK2/9UC7t4RUAH5wnmPgEjmkajQ6wTAyJ/ne2wl6YLAAAAQQDMGlHb#U+d7t25a33dAQjzu+ynQQnz5JED/2ve4bnEzzc4KAc+IZL9IszNFvK/cxtwX#FVmyHrEDOf1qgL7nRrRX#====#
onie_initargs=setenv bootargs quiet console=$consoledev,$baudrate maxcpus=4 pci=pcie_bus_safe cpuidle.off=1
onie_machine=accton_as4224
onie_machine_rev=0
onie_platform=arm64-accton_as4224-r0
onie_platformargs=setenv bootargs $bootargs serial_num=${serial#} eth_addr=$ethaddr $onie_bootargs $onie_debugargs 
onie_rescue=setenv onie_boot_reason rescue && boot
onie_start=0x00210000
onie_sz.b=0x00df0000
onie_uninstall=setenv onie_boot_reason uninstall && boot
onie_update=setenv onie_boot_reason update && boot
onie_vendor_id=259
platform=accton_as4224
ramdisk_addr_r=0x8000000
ramfs_name=-
root=root=/dev/nfs rw
rootpath=/srv/nfs/
serial#=511448X2042112
serverip=0.0.0.0
set_bootargs=setenv bootargs $console $root ip=$ipaddr:$serverip:$gatewayip:$netmask:$hostname:$netdev:none nfsroot=$serverip:$rootpath,tcp,v3 $extra_params $cpuidle
soc=mvebu
stderr=serial@512000
stdin=serial@512000
stdout=serial@512000
usb_pgood_delay=5000
vendor=Marvell
onie_version=2020.02.00.09
ver=onie_uboot_version=U-Boot 2019.10-10.0.0-gef8ade1 (Nov 06 2020 - 17:47:36 +0800) - 10.3.3.0.1
onie_uboot_version=onie_uboot_version=U-Boot 2019.10-10.0.0-gef8ade1 (Nov 06 2020 - 17:47:36 +0800) - 10.3.3.0.1
onie_build_date=2020-11-06T15:15+08:00
onie_bootcmd=echo Loading Open Network Install Environment ...; echo Platform: $onie_platform ; echo Version : $onie_version ; sf probe && sf read $loadaddr $onie_start ${onie_sz.b} && run onie_args && bootm ${loadaddr}
boot-config-default=TkVUREVWPW1hMQpCT09UTU9ERT1JTlNUQUxMRUQKU1dJPWltYWdlczo6bGF0ZXN0Cg==
onl_installer_initrd=onl-loader-fit.itb:2772ffc:355205a
nos_bootcmd=setenv onl_loadaddr 0x20000000; setenv onl_platform arm64-accton-as5114-48x-r0; setenv onl_itb arm64-accton-as5114-48x-r0.itb; setenv bootargs $console onl_platform=$onl_platform arm-smmu.disable_bypass=n maxcpus=4 pci=pcie_bus_safe cpuidle.off=1 log_buf_len=4M initcall_debug; scsi reset; ext2load scsi 0:1 $onl_loadaddr $onl_itb; bootm $onl_loadaddr#$onl_platform
```

paulmenzel, 2021-04-07:
/cc @sonoble, @buczek

sonoble, 2021-04-07:
@richardlee66 can possibly chime in here.

ehdoyle, 2021-12-30:
Hi - just checking in to see if there was any progress on this. Unless the manufacturer upstreams the code there isn't much to be done, and I'd close this out as an "unsupported platform" in the next few days.  Thanks!