opencomputeproject/onie #971: make: *** No rule to make target 'conf/crosstool/gcc-4.9.2/uClibc-ng-1.0.38/crosstool.x86_64.config', needed by '/home/build/src/onie/build/x-tools/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.38/build/.config

AlirezaPhD, 2021-10-01:
I followed the instruction: https://github.com/opencomputeproject/onie/tree/master/contrib/build-env#readme
to create image for the new switch.
But when I run:    
build@8d8b3f116ee2:~/src/onie/build-config$ make -j1 MACHINEROOT=../machine/accton MACHINE=accton_wedge100bf_32qs all

I faced with this error : 
_make: *** No rule to make target 'conf/crosstool/gcc-4.9.2/uClibc-ng-1.0.38/crosstool.x86_64.config', needed by '/home/build/src/onie/build/x-tools/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.38/build/.config'.  Stop._

Any help would be helpful. Thanks in advance.
all the logs are here:
build@8d8b3f116ee2:~/src/onie/build-config$ make -j1 MACHINEROOT=../machine/accton MACHINE=accton_wedge100bf_32qs all
mkdir: created directory '/home/build/src/onie/build/images'
mkdir: created directory '/home/build/src/onie/build/download'
mkdir: created directory '/home/build/src/onie/build/accton_wedge100bf_32qs-r0'
mkdir: created directory '/home/build/src/onie/build/accton_wedge100bf_32qs-r0/stamp'
mkdir: created directory '/home/build/src/onie/build/accton_wedge100bf_32qs-r0/initramfs'
==== Getting Linux ====
  Downloading linux-4.9.95.tar.xz from: http://mirror.opencompute.org/onie/linux-4.9.95.tar.xz
2021-10-01 10:03:30 URL:http://mirror.opencompute.org/onie/linux-4.9.95.tar.xz [93289700/93289700] -> "/home/build/src/onie/build/download/linux-4.9.95.tar.xz" [1]
  Got          linux-4.9.95.tar.xz

linux-4.9.95.tar.xz: OK
==== Extracting Linux ====
==== patching  Linux ====
'/home/build/src/onie/machine/accton/kernel/driver-intel-igb.patch' -> '/home/build/src/onie/build/accton_wedge100bf_32qs-r0/kernel/patch/driver-intel-igb.patch'
'/home/build/src/onie/machine/accton/kernel/driver-igb-broadcom-phy.patch' -> '/home/build/src/onie/build/accton_wedge100bf_32qs-r0/kernel/patch/driver-igb-broadcom-phy.patch'
Initialized empty Git repository in /home/build/src/onie/build/accton_wedge100bf_32qs-r0/kernel/linux-4.9.95/.git/
Checking for changes in the working directory ... done
Importing patch "driver-intel-igb.patch" ... done
Importing patch "driver-igb-broadcom-phy.patch" ... done
Now at patch "driver-igb-broadcom-phy.patch"
==== Copying conf/kernel/4.9.95/linux.x86_64.config to /home/build/src/onie/build/accton_wedge100bf_32qs-r0/kernel/linux/.config ====
'conf/kernel/4.9.95/linux.x86_64.config' -> '/home/build/src/onie/build/accton_wedge100bf_32qs-r0/kernel/linux/.config'
==== Merging patches from /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config to /home/build/src/onie/build/accton_wedge100bf_32qs-r0/kernel/linux/.config ====
Using /home/build/src/onie/build/accton_wedge100bf_32qs-r0/kernel/linux/.config as base
Merging /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config
Value of CONFIG_IGB is redundant by fragment /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config:
Value of CONFIG_USB_XHCI_HCD is redundant by fragment /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config:
Value of CONFIG_USB_XHCI_PLATFORM is redundant by fragment /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config:
Value of CONFIG_USB_HID is redefined by fragment /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config:
Previous value: # CONFIG_USB_HID is not set
New value: CONFIG_USB_HID=y

Value of CONFIG_HIDRAW is redefined by fragment /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config:
Previous value: # CONFIG_HIDRAW is not set
New value: CONFIG_HIDRAW=y

Value of CONFIG_E1000 is redefined by fragment /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config:
Previous value: CONFIG_E1000=y
New value: # CONFIG_E1000 is not set

Value of CONFIG_E1000E is redefined by fragment /home/build/src/onie/machine/accton/accton_wedge100bf_32qs/kernel/config:
Previous value: CONFIG_E1000E=y
New value: # CONFIG_E1000E is not set

#
# merged configuration written to /home/build/src/onie/build/accton_wedge100bf_32qs-r0/kernel/linux-4.9.95/.config (needs make)
#
make: *** No rule to make target 'conf/crosstool/gcc-4.9.2/uClibc-ng-1.0.38/crosstool.x86_64.config', needed by '/home/build/src/onie/build/x-tools/x86_64-g4.9.2-lnx4.9.95-uClibc-ng-1.0.38/build/.config'.  Stop.

ehdoyle, 2021-10-01:
Hi - that platform hasn't been updated to work with the upgraded code in master -  try building off the 2021.08.br branch of ONIE.
(And to be fair, the documenation needs to be upgraded to make this more obvious.)
There is a matrix of switches, build releases and build environments here:
[onie-build-targets.json](https://github.com/opencomputeproject/onie/blob/master/build-config/scripts/onie-build-targets.json)

AlirezaMohammadpour85, 2021-10-01:
Thank you for your answer. Yes I tried the previous versions. The one for 2020.11 seems to work.

edrozenberg, 2022-08-21:
Helped me also thanks. Had to use `2020.05br` for my `cel_e1031` hardware. Would be awesome to see this hint in the documentation, as I was completely stuck.