opencomputeproject/onie #978: EDA-DIAG help requested ... I have so much to learn

Z4Boz, 2021-12-29:
Hello, hope this finds you well...

Has anyone out there used the EDA-DIAG tools part of ONIE to diagnose a network switch before?

I have spent a long time learning to install it (I am a linux total newbie) --- but have successfully installed ONIE on a blank hard drive, and the DIAG-OS ---- and now, the EDA-DIAG tools package. 

But I am struggling with the commands on how to run it...  there is so little documentation on this subject --- I was hoping someone with experience could kindly help me out....

Thanks,
Z

ehdoyle, 2021-12-30:
Hi - I'd suggest posting this on the [ONIE mailing list](https://ocp-all.groups.io/g/OCP-ONIE) - that would be the best place to see what people's experiences have been.
A quick search of the  [topics](https://ocp-all.groups.io/g/OCP-ONIE/topics) doesn't seem to show anything, so I'd agree this is a topic that could use more documenatation.

Z4Boz, 2021-12-31:
I’ll try that —- there’s very little in general on this topic so thnx did your help 
    	Get Outlook for iOS

Z4Boz, 2022-01-04:
Hi ---  I did as you suggested… emailed the ONE mailing list.  So far, no replies… 

 

Quick question:  Is it my imagination or does dell “lock down” on inquiries into their documentation on network oriented forums?

 

Is that the reason it’s so hard to find posts related to their switches / troubleshooting?  

 

I’ve had several posts removed from admins…. 

 

-Z

 

🂡🂸

ehdoyle, 2022-07-17:
Hi - just got back to this while doing some cleanup.
Yeah, the ONIE mailing list is a bit quiet these days - everybody seems to be happy or tuned out - I'm not sure which :)

I'm not sure about Dell documentation issues - this is the first I've heard about it, but I'll keep an eye out for similar stories.