opencomputeproject/onie #990: SFTP is more reliable, when will the SFTP function be added?

wcchen0519, 2022-03-11:
TFTP has no access authorization and authentication mechanism;
Secure File Transfer Protocol (SFTP) is a Secure File Transfer Protocol. SFTP uses encryption and decryption technologies to ensure the security of transferred files;

so SFTP is more reliable, when will the SFTP function be added?

ehdoyle, 2022-03-14:
Hi - currently nobody is working on SFTP, but it would be great to have that added to ONIE.  As an open source project, ONIE has thrived and evolved through the contributions of the community, so we very much encourage the submission of pull requests to improve the code base.
Thanks!

ehdoyle, 2022-07-17:
Closing out due to inactivity.

closed 2022-07-17 by ehdoyle