opencomputeproject/onie #992: error while "make MACHINE=kvm_x86_64 -j4 all demo"

zhengbiqing, 2022-03-20:
Hi
I built onie for kvm_x86_64, but there is some error:

popthelp.c: In function 'poptPrintHelp':
popthelp.c:666:9: warning: variable 'xx' set but not used [-Wunused-but-set-variable]
     int xx;
         ^~
poptint.c:12:28: warning: 'utf8_skip_data' defined but not used [-Wunused-const-variable=]
 static const unsigned char utf8_skip_data[256] = {
                            ^~~~~~~~~~~~~~
test1.c:164:35: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
    { "bits", '\0', POPT_ARG_BITSET|POPT_ARGFLAG_DOC_HIDDEN, &aBits, 0,
                                   ^
libtool: install: warning: remember to run `libtool --finish /usr/lib'
configure: WARNING: zlib.h: accepted by the compiler, rejected by the preprocessor!
configure: WARNING: zlib.h: proceeding with the compiler's result
make[1]: warning: -j1 forced in submake: resetting jobserver mode.
libtool: warning: remember to run 'libtool --finish /usr/lib'
kexec/arch/i386/crashdump-x86.c: In function 'get_kernel_page_offset':
kexec/arch/i386/crashdump-x86.c:55:6: warning: unused variable 'kv' [-Wunused-variable]
  int kv;
      ^~
x86_64-onie-linux-uclibc-ar: creating libutil.a
Can't open /home/zbq/onie-git-bak/encryption/machines/kvm_x86_64/keys/ONIE/efi-keys/ONIE-shim-key-cert.der for reading, No such file or directory
140164733244544:error:02001002:system library:fopen:No such file or directory:../crypto/bio/bss_file.c:69:fopen('/home/zbq/onie-git-bak/encryption/machines/kvm_x86_64/keys/ONIE/efi-keys/ONIE-shim-key-cert.der','rb')
140164733244544:error:2006D080:BIO routines:BIO_new_file:no such file:../crypto/bio/bss_file.c:76:
unable to load certificate
make: *** [make/shim.make:115: /home/zbq/onie-git-bak/build/user/x86_64-g8.3.0-lnx5.4.86-uClibc-ng-1.0.38/stamp/shim-build] Error 1**
make: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '/home/zbq/onie-git-bak/encryption/machines/kvm_x86_64/keys/ONIE/efi-keys/ONIE-shim-key-cert.pem', needed by 'certs/signing_key.x509'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1732: certs] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [make/kernel.make:116: /home/zbq/onie-git-bak/build/kvm_x86_64-r0/stamp/kernel-build] Error 2
x86_64-onie-linux-uclibc-ar: creating apps/libapps.a
x86_64-onie-linux-uclibc-ar: creating libssl.a
x86_64-onie-linux-uclibc-ar: creating test/libtestutil.a
x86_64-onie-linux-uclibc-ar: creating libcrypto.a

Anyone can help me? Thanks!

ehdoyle, 2022-03-22:
Hi - the kvm_x86_64 build defaults to a Secure Boot configuration, and is looking for both keys, and a signed shim (ONIE can build both).
If you do not want secure boot, you can:
# use a kernel configuration that does not expect signing
cp machine/kvm_x86_64/kernel/config-insecure machine/kvm_x86_64/kernel/config
# edit machine/kvm_x86_64/machine.make, and set:
SECURE_BOOT_ENABLE = no
SECURE_BOOT_EXT = no
SECURE_GRUB = no
And comment out:
MACHINE_SECURITY_MAKEFILE
Then run a 'make distclean' and rebuild to make sure everything reset clearly.

If you're trying out Secure Boot, you'll need to
# Generate the keys
`make signing-keys-generate`   ( or 'make signing-keys-install' if you want them added to the virtual USB drive used at emulation run time)
# create a self signed shim
`make shim-self-signed`
# and then run the kvm build
`make kvm_x86_64 all demo -j4`

There is also a video tutorial here for Secure Boot
[https://www.youtube.com/watch?v=evzkiiRRIvw
](https://www.youtube.com/watch?v=evzkiiRRIvw
)
I guess the documentation could use some updating. I'd thought these instructions were already in there, but I didn't find them...
Hope that helps.

zhengbiqing, 2022-03-22:
Thanks very much!

 

 
 
 
 
 
 
 
 
 
 
 
 
 https://www.youtube.com/watch?v=evzkiiRRIvw

closed 2022-03-22 by zhengbiqing