opencomputeproject/onie #993: Facing issue While building and loading KVM_X86_64 machine ONIE iso

ankitkumarsr, 2022-03-25:
There are couple of package not available or mentioned example:
python-syphinx.
There are two type of build image secure and insecure.
for Secure build some step are missing in build instruction.
Step followed :(running build on virtual box Ubuntu 20.04.02)

$ git clone https://github.com/opencomputeproject/onie.git
$ cd build-config
$ sudo apt-get update
$ sudo apt-get install build-essential
$ make debian-prepare-build-host
$ make  MACHINE=kvm_x86_64 all recovery-iso

build failed couple of time asking for signed certificate.
$ sudo apt-get install python3.8-dev tree
$ make  MACHINE=kvm_x86_64 shim-self-sign
$  make signing-keys-generate MACHINE=kvm_x86_64

For In-secure build 
1.Copy  machine/kvm_x86_64/kernel/config-insecure to machine/kvm_x86_64/kernel/config
2. Modified machine.make inside folder machine/kvm_x86_64 as below
```
git diff machine/kvm_x86_64/machine.make
diff --git a/machine/kvm_x86_64/machine.make b/machine/kvm_x86_64/machine.make
index 438d762e..804cf342 100644
--- a/machine/kvm_x86_64/machine.make
+++ b/machine/kvm_x86_64/machine.make
@@ -64,15 +64,15 @@ SKIP_ETHMGMT_MACS = yes
 #  that this provides.
 # The kernel/config-insecure file is provided as
 #  an example.
-SECURE_BOOT_ENABLE = yes
+SECURE_BOOT_ENABLE = no
 
 # Enable extended secure boot:
 #  Activates - ONIE password
-SECURE_BOOT_EXT = yes
+SECURE_BOOT_EXT = no
 `>`
 # Enable GRUB verification of files and passwords
 # Requires secure boot
-SECURE_GRUB = yes
+SECURE_GRUB = no

``` 
 After Successful build got image inside build/images
Issued facing while running mk-vm.sh file

```
Trying 127.0.0.1...
telnet: Unable to connect to remote host: Connection refused
to kill kvm:  sudo kill 9567
```

ehdoyle, 2022-03-26:
Hi - the host build configuration should work on most systems, but given the differences (both known and unknown) in individual deployments it gets difficult to provide support remotely, and debug usually ends with either:  "well, it works on my system " or "that configuration is not officially supported" (That said, pull requests for proved working configurations are always welcome).
Of course none of that is helpful.

To get around the "works for me" problem, I'd suggest using the DUE ONIE Docker container to create a Debian 10 build environment - which should produce exactly the same build environment that was used for development and releases.
The .deb here: [https://github.com/CumulusNetworks/DUE/releases/tag/v3.1.0](https://github.com/CumulusNetworks/DUE/releases/tag/v3.1.0) should install on an Ubuntu 20 system with:
```
sudo dpkg -i due_3.1.0-1_all.deb
sudo apt install --fix-broken
```
(The --fix-broken will add any dependencies like Docker that are not already on your system. You'll have to add yourself to the Docker group and log in again, but DUE will walk you through that when it runs)
Or you can check out the source and run directly from DUE's directory, if you prefer.

The instructions for setting up an ONIE Docker image can be found here:
[https://github.com/CumulusNetworks/DUE/blob/master/templates/onie/README.md](https://github.com/CumulusNetworks/DUE/blob/master/templates/onie/README.md)

For running the virtual machine, try using the scripts under the onie/emulation directory.
Details are here:
[https://github.com/opencomputeproject/onie/blob/master/emulation/README-onie-vm.md](https://github.com/opencomputeproject/onie/blob/master/emulation/README-onie-vm.md)

Putting it all together, there is a video tutorial for the secure boot process here:
[https://www.youtube.com/watch?v=evzkiiRRIvw](https://www.youtube.com/watch?v=evzkiiRRIvw)
which demonstrates the emulation scripts. For an insecure boot, the key management steps should be okay to ignore.
At that point you'll be using what should be a known working configuration that can be reproduced and debugged remotely if you are still seeing issues.
Let me know how it goes.

ehdoyle, 2022-07-17:
Closing out due to inactivity.

closed 2022-07-17 by ehdoyle