opencomputeproject/onie #1000: Can we Add new CPU

ankitkumarsr, 2022-06-02:
I want to integrate ONIE to intel denverton C3508 , IS there any procedure to add new cpu?

ehdoyle, 2022-06-05:
I don't think anything is defined as such.  You can check the existing codebase for any chip specific modifications though.
Dumping the git commit log and grepping for 'denverton' it looks like a number of platforms support the C35**3**8, and these platforms in particular support the C35**0**8 
[Alpha Networks SPX70D0-168F](https://github.com/opencomputeproject/onie/blob/013f9e74fe54d00d615a2bbb3b1740a4146097ef/build-config/scripts/onie-build-targets.json#L111)
[Ufispace S9501-18SMT](https://github.com/opencomputeproject/onie/blob/013f9e74fe54d00d615a2bbb3b1740a4146097ef/build-config/scripts/onie-build-targets.json#L257)
Hope that helps.

ankitkumarsr, 2022-06-14:
while integrating Onie to custom board through below error
[image]

Please provide help on this issue

ankitkumarsr, 2022-07-10:
For my custom-board i am able to get onie installation option attached in (Onie-image2.png)
[image]

if booted manually , the following expected menu is not coming
[image]

Please provide suggestion What is missing here to get desired output.

ehdoyle, 2022-07-17:
Hi - This seems to be implementation specific issue, it's not really an actionable bug in ONIE, so I'll be moving to close this out shortly.

I'd suggest examining the ONIE builds of the previously mentioned systems to see if they are doing anything differently.

Apart from that, the ONIE mailing list:
[https://ocp-all.groups.io/g/OCP-ONIE](https://ocp-all.groups.io/g/OCP-ONIE) 
can also be a useful resource, as someone else may have encountered similar issues.

When posting there, (or seeking help with any issue ) I recommend providing as much information as possible about:
- What you expect to be happening ( Ex: ONIE Grub menu should come up after embed )
- Why you expect that ( Ex: ONIE embedding worked, system rebooted, the menu was the next step )
- What is happening ( Ex: ONIE grub menu does not come up, and the last output is...? )
- What you know about what is happening ( Ex: debug output, trace logs, events before the failure, exact steps taken to reproduce the problem )
- What you have done so far to fix the problem that has not worked ( Ex: used set -x in setup scripts to observe installation, booted rescue mode to examine the deployed Grub configuration, turning it off and on again, etc ). 

Showing your work so far is **very** important for a few reasons:

1.  It will reduce the amount of time required to get to a solution (since they know what you have tried, they will not waste time asking you to do things you already have done.
2. Someone may be able to diagnose the problem from information provided, and can quickly respond, rather than deciding that it would be too much work to figure out.
3. It also demonstrates that you are actively trying to solve the problem and are not expecting others to do your work for you. You want to be seen as trying to solve the problem. 

That said, the ONIE mailing list is pretty quiet these days, and the problem you are encountering seems to be more of an issue with your particular implementation than ONIE, so you may not get a response.

Hope that helps.

saikumar25, 2022-07-19:
1) We made a bootable USB by using recovery image built for KVM
2) While trying to boot from the USB, the first screen which pops is a black screen with just "Welcome to GRUB !" text.
3) Within some time , kernel boot messages start appearing and in some time , ONIE prompt comes up in installer mode .
4) When trying to install a NOS , we are getting below error :
[image]
5)Also in the tutorials we've been through until now , during initial/consequent ONIE boots we should be able to see the menu coming up something like this (Embed ONIE and Rescue ONIE only in case of initial boot) :
[image]
6)We are unable to see any grub menu at all(not getting ONIE embedding menu or ONIE installer menu) , during boot up .