opencomputeproject/onie #999: make[2]: *** No rule to make target '/home/alexander/ONIE/onie/encryption/machines/kvm_x86_64/keys/ONIE/efi-keys/ONIE-shim-key-cert.pem', needed by 'certs/signing_key.x509'.  Stop.

ping2alexander, 2022-06-02:
alexander@alexander-Latitude-3420:~/ONIE/onie/build-config$ make MACHINE=kvm_x86_64 all recovery.iso
==== Building cross linux ====
make[1]: Entering directory '/home/alexander/ONIE/onie/build/kvm_x86_64-r0/kernel/linux-5.4.86'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND  objtool
  CHK     include/generated/compile.h
make[2]: *** No rule to make target '/home/alexander/ONIE/onie/encryption/machines/kvm_x86_64/keys/ONIE/efi-keys/ONIE-shim-key-cert.pem', needed by 'certs/signing_key.x509'.  Stop.
make[1]: *** [Makefile:1732: certs] Error 2
make[1]: Leaving directory '/home/alexander/ONIE/onie/build/kvm_x86_64-r0/kernel/linux-5.4.86'
make: *** [make/kernel.make:116: /home/alexander/ONIE/onie/build/kvm_x86_64-r0/stamp/kernel-build] Error 2
alexander@alexander-Latitude-3420:~/ONIE/onie/build-config$

ehdoyle, 2022-06-05:
Hi - just checking if this is still an issue? The strikethrough in the the text implies that it isn't.
If it is still an issue, can you supply the build commands used from code checkout to the point where the failure occurred, and describe the build environment being used?
Otherwise if you solved the issue  I can just close this out.  Thanks!

ehdoyle, 2022-06-25:
Closing out after 20 days.

closed 2022-06-25 by ehdoyle