opencomputeproject/onie #1008: Ubuntu 20.04 full Offline build like onie/contrib/debian-iso/ using cook-bits.sh

mariusz-stordis, 2022-08-04:
Is it possible to build a fully offline installer (.bin file) with this script. I would like to ask someone experienced for guidance.

https://github.com/opencomputeproject/onie/tree/master/contrib/debian-iso
https://github.com/wililupy/ubuntu-onie

mariusz-stordis, 2022-10-11:
> Hi - I'd tried replying to your email but it kept bouncing.
> Sorry to say I've never tried this - the code has been around in ONIE since before I became project lead, so what's there is pretty much the extent of it,
> Plus it's pretty old, so it may make sense to use an older Ubuntu iso to try it out.
>
> It looks like it's using the standard iso image, but I've not dug in to how that's constructed. There may be parallels between that and how ONIE constructs an installer image - but again I've never looked at the code or tried installer creation outside of what ONIE does.
>
> Sorry that's not much to go on.

Thank you. Do you know someone who could help with this? I've seen something like onie-packer on github works offline but it uses onie's built-in kernel (version 4) and that one doesn't support Ubuntu 20. Alternatively, is there any command that will do an ONIE kernel upgrade?
Przetłumaczono za pomocą DeepL https://www.deepl.com/app/?utm_source=android&utm_medium=app&utm_campaign=share-translation

Mariusz Szlęzak
Support & Service Engineer

[cid:Stordis_Logo_Flat+slogan_b5d4399f-227e-4183-b97f-e322351edbfd.png]

E: ***@***.***  STORDIS GmbH
| M: +48 666 914 823    Rosenwiesstr. 17 | 70567 Stuttgart | DE
Teams: Chat/Call via ***@***.***>    T: +49 (0)711 342158 0
[cid:icon-linkedin_191f7fb5-9be0-4791-9d27-d5bbc07d1537.png]<https://www.linkedin.com/company/1443466>[cid:icon-twitter_ff6068fb-5511-438d-aa97-f35ed48dd58a.png]<https://twitter.com/STORDIS_GmbH>[cid:icon-facebook_e6f8d856-cd1d-4bc7-a358-4472c097274c.png]<https://www.facebook.com/STORDIS-GmbH-276466799779934>[cid:icon-youtube_671ddf3e-e953-4442-9a77-f9e79e4a6678.png]<https://www.youtube.com/channel/UCju-Xk7eWf-j0vu-RYkJwsA>     www.stordis.com<http://www.stordis.com> | www.stordirect.com<http://www.stordirect.com/>

<https://stordis.com/blog/open-networking/switch2open-new-product-launch-transceivers-dac-fiber-optic-cables/?utm_campaign=switch2open-launch&utm_source=mail-signature&utm_medium=email>[cid:banner-email-switch2open_328c31f7-bc59-4ba1-a30a-e6b3ad22f1a7.jpg]<https://stordis.com/blog/open-networking/switch2open-new-product-launch-transceivers-dac-fiber-optic-cables/?utm_campaign=switch2open-launch&utm_source=mail-signature&utm_medium=email><https://stordis.com/blog/open-networking/introducing-the-stordis-switch-services-service-sessions-with-a-qualified-network-engineer/?utm_source=mail-signature&utm_campaign=switch-service&utm_medium=email>

Are you satisfied with our performance? Please help us improve and evaluate us here.<https://stordis.com/customer-survey/?utm_campaign=customer-survey&utm_source=mail-signature&utm_medium=email> <https://stordis.com/blog/open-networking/switch2open-new-product-launch-transceivers-dac-fiber-optic-cables/?utm_campaign=switch2open-launch&utm_source=mail-signature&utm_medium=email>

Court of Jurisdiction: Stuttgart, HRB 724261, Management: Waldemar Scheck.
You can find information on how and why we store personal data in our data protection declaration<https://stordis.com/data-protection-declaration/>.

This message is private and confidential. If you have received this message in error, please notify us and remove it from your system. If you are not the intended recipient you are notified that disclosing, copying, distributing or taking any action in reliance on the contents of this information is strictly prohibited. Any views or opinions expressed in this email do not necessarily represent those of the company

ehdoyle, 2022-10-12:
Hi - sorry to say I'm not aware of anyone working on this. You might try the ONIE mailing list [https://ocp-all.groups.io/g/OCP-ONIE](https://ocp-all.groups.io/g/OCP-ONIE) to see if anyone has some insight.
Insofar as kernel upgrade goes, I don't think there's been much work done on that - typically a new version of ONIE is built and everything upgrades at once.
For what that's worth?