opencomputeproject/onie #1009: Any way to use newer ONIE on old stuff (cel_e1031)

edrozenberg, 2022-08-24:
Based on matrix rec:
https://github.com/opencomputeproject/onie/blob/master/build-config/scripts/onie-build-targets.json

I built `2020.05br` using Debian v9, for my Super Micro (SMC) G3648 aka Celestica 1031.

Is/will there be a way to build recent releases for these old switches? Os is 2020 the end of the road?

I ask because the 2020 ONIE does not support some niceties like ext4, or FS csums. And would be nice to be able to use a recent Debian or recent build env, instead of downloading old Debian stuff.

I also had an issue installing an old SONiC 2019 build, it borked ONIE/Grub, and I had to use the onie restore image to get onie working again. This might be an issue with SONiC though, no the old ONIE.

Anyway in summary: is there a way to build a recent ONIE for old crap switches? Or don't bother? Thanks!

ehdoyle, 2022-08-25:
Hi - the short answer is upgrades are absolutely possible, but they require someone putting in the effort to resolve component upgrade issues (as the latest tool chain is more strict about errors) and having the resources to test the changes, since we all know getting it to build and getting it to run correctly are two different things.
Typically it falls on the manufacturer to update their version of ONIE for a particular switch, as they'd be in the best position to do so, but there hasn't seemed to be much activity on that front.  The versioned build environments and releases detailed in the onie-build-targets.json file were a way to allow ONIE to upgrade without breaking builds that were already working, but  most (all?) of the updates there have been for new platforms, rather than upgrades of existing ones.
For an example what is involved in the upgrade process, I'd say take a look at the changes for the x86 and arm64 virtual platforms, as they went through all of that.

edrozenberg, 2022-08-25:
@ehdoyle I understand, old stuff gets orphaned by lots of projects, it's the way of the world. I just wanted these old switches for a home lab to use with SONiC. None of the prebuilt SONiC builds installed on these - I'll try building by own SONiC and if that fails, I'll chuck this hardware. Too many challenges dealing with it.

closed 2022-08-25 by edrozenberg