opencomputeproject/onie #1010: A mention of the base MAC address Tlv offset on the website is wrong

miquelraynal, 2022-09-22:
On the following page: https://opencomputeproject.github.io/onie/design-spec/hw_requirements.html
In the Type Code Values array: https://opencomputeproject.github.io/onie/design-spec/hw_requirements.html#type-code-values

Offset 0x2A specifies:

> A two-byte big-endian unsigned integer describing the number of sequential MAC addresses allocated to this device, starting with the value specified in the MAC #1 Base TLV (code 0x2A). Valid values for this field range from 1 to 65535.

While it should be:

> A two-byte big-endian unsigned integer describing the number of sequential MAC addresses allocated to this device, starting with the value specified in the MAC #1 Base TLV (code 0x24). Valid values for this field range from 1 to 65535.

(the "MAC #1 Base TLV" code should be 0x24 instead of 0x2A)

ehdoyle, 2022-10-06:
Thanks for pointing that out - that typo has been in there _quite_ a while.  So just to clarify, data ( for, say 2 MAC addresses) should look like this in memory.
0x2a 0x00 0x02
0x24 0x12 0x34 0x56 0x78 0x9A 0xBC
0x24 0x12 0x34 0x56 0x78 0x9A 0xBD

Rather than what is written, which would look like:
0x2a 0x00 0x02
0x2a 0x12 0x34 0x56 0x78 0x9A 0xBC
0x2a 0x12 0x34 0x56 0x78 0x9A 0xBD
...which would be interpreted as an extremely large number of MAC addresses, without providing any.

So the proposed fix is to change:
A two-byte big-endian unsigned integer describing the number of sequential MAC addresses allocated to this device, starting with the value specified in the MAC #1 Base TLV (code 0x2**A**). Valid values for this field range from 1 to 65535.

To
A two-byte big-endian unsigned integer describing the number of sequential MAC addresses allocated to this device, starting with the value specified in the MAC #1 Base TLV (code 0x2**4**). Valid values for this field range from 1 to 65535.

Correct?

miquelraynal, 2022-10-07:
Yes!