opencomputeproject/onie #1014: Size of ONIE?

tchandramohan, 2022-10-14:
Hi Alex,

What is the size of the filesystem where ONIE is installed successfully on? 
(I did not find a concrete answer to this in the documentation)
(Not the iso file or demo NOS, but the size ONIE takes up for it's instalation)

ehdoyle, 2022-10-14:
Hi,
 I'd say about 10-20MB seems to be about right. It will vary some depending on the target you're building for, (since the versions of software components and tools used to build them are going to vary) and how you want to measure it ( size of the installer, space the installer asks for, space the binaries actually take up, etc ). If you need better resolution than that, the best way to know for your particular case is to build and run it, then measure in a way that makes sense for your application -  which I'm rather curious about. Given the storage capacities of most modern devices, I would think using 20MB would hardly be noticed.
 If you don't have that hardware handy, you can build and run the x86 and arm64 target machines if you have QEMU installed  and measure things that way (again ,the emulation environments come through for ease in prototyping), to get a better idea of what you would be in for.
 HTH

tchandramohan, 2022-10-25:
Thank you! Closing this ticket.

closed 2022-10-25 by tchandramohan