opencomputeproject/onie #1015: Missing tar file | parted-3.1.tar.xz

tchandramohan, 2022-10-18:
Hi Alex,

The build for a KVM target was successful. However when I run a distclean and try to build ONIE & Demo NOS for a KVM target, the build fails at one point. In particular, the build says it is unable to find the file -> parted-3.1.tar.xz in the directory src/onie/build/download

However, in the same directory src/onie/build/download, I find the file 'parted-3.4.tar.xz'
I am attaching screenshots for both. 
Please tell me how to proceed. Is this a bug? 

[image]
[image]

ehdoyle, 2022-10-18:
Hi - had you also building the ARM virtual target? I recall that the ARM and x86 virtual builds used two different versions of one package and building one after the other in the same build area would fail. (IIRC upgrading them both to the latest version was a dependency cascade nightmare that wasn't worth the cost of addressing at the time.)
If that is the case I'm surprised a distclean didn't take care of that. Emptying out the ./build area should fix that, or you may be able to selectively delete the offending downloaded parted tar.xz file and a 'make all' should redownload it.

tchandramohan, 2022-10-18:
It looks like an ARM target image was in place under the build/images directory. 
I did a distclean and emptied the build area like you suggested. 
I have ONIE installer files and demo NOS files created under onie/build/images
[image]

Question: Currently, I am following this readme for ONIE installation [https://github.com/opencomputeproject/onie/blob/master/machine/kvm_x86_64/INSTALL#L72]
However this readme does not highlight how to install demo NOS since the installer is a .bin file. 

Previously, I was able to successfully install ONIE (for a KVM target) using this setup
[image]
And I was also able to boot from hard disk after instalation.

How do I get demo NOS in after installing ONIE?

ehdoyle, 2022-10-18:
Once ONIE is embedded on the hard drive, it should default to running a waterfall discovery process to find images to install, and while there are a lot of places one could put the image to be found,  I usually copy the demo os bin file to be under a web server, then for onie:
onie-stop  -- to stop the discovery process
onie-nos-install  --URL to the demo OS image.
...at which point it should install, reboot, and come up in the demo os.

closed 2022-10-25 by tchandramohan