opencomputeproject/onie #1018: Customize ONIE?

tchandramohan, 2022-10-25:
Hi Alex,

I built and installed ONIE & demo OS for a kvm_x86_64 target. I also checked out armv8a target to study the build files. 
Next, I have an evaluation board with me. How should I go about installing ONIE and demo OS on this board?
This board (or the chip) is not listed under onie/machines.
In that case, ONIE should be customized to the target platform. 
Can you please guide?

ehdoyle, 2022-10-25:
You'll want to start with a build that is close to your target platform. I'd suggest building the installer iso for that and then trying to boot your platform off of it. Ideally you get some output from the boot that will indicate where support needs to be added. 
You may also want to check ONIE's git history, or closed bugs for hints on problems you'll encounter as something similar may have come up in the past.  There's also the ONIE mailing list [https://ocp-all.groups.io/g/OCP-ONIE](https://ocp-all.groups.io/g/OCP-ONIE), which, while it's been rather quiet, might have somebody who has insight in to the particulars of your situation.

tchandramohan, 2022-10-27:
Thanks for your response. 
Your advice is more reverse engineering. 
I will check it out.
In the meantime, I was hoping for more of a straight forward approach towards building and installing ONIE for my target. 
Any thoughts on this, Alex?

ehdoyle, 2022-10-28:
I'd say try the mailing list and see if anybody has anything else to add. There's also some general guidelines in the porting guide here:
[https://opencomputeproject.github.io/onie/developers/porting.html](https://opencomputeproject.github.io/onie/developers/porting.html)

tchandramohan, 2022-11-03:
Alright. Thank you, Alex.
Question: Does ONIE use a standard u-boot or a modified u-boot? (Any links to documentation about this?)

ehdoyle, 2022-11-03:
It should work with a standard u-boot, although there's not much in the documentation about that.
I'd suggest looking at the arm platforms - there's a list in [onie-build-targets.json](https://github.com/opencomputeproject/onie/blob/master/build-config/scripts/onie-build-targets.json) to see how they're handling things.

tchandramohan, 2022-11-03:
Thank you, Alex. 
Following the above question, I have another question surrounding ONIE & u-boot.
Looking forward for your insight on the same. 

Question: After ONIE is installed successfully in the NOR flash (or emmc perhaps), how do we configure u-boot such that we boot directly into demo OS?

In specific, I want to know how can we influence u-boot:
a] To boot into ONIE 
b] To boot into demo OS

ehdoyle, 2022-11-03:
That may depend on the platform. I've seen two approaches used.
One uses u-boot to load the kernel, initrd and device tree blob and boots the OS directly - so the ONIE installer has to be aware enough of how the platform works to do this.
I've also seen things configured such that u-boot ultimately loads grub, and then boot proceeds from there. So grub can be part of the boot process, but doesn't have to be. ARM is a bit trickier than x86 as the hardware tends to be less standardized.