opencomputeproject/onie #1027: Error building ONIE image for machine quanta_common_rglbmc

luigitalboy, 2023-03-31:
Hello, 
I'm trying to build a ONIE image for a Quanta Mesh T4048-IX8 but I receive this error 

onie@oniebuilder2:~/onie/build-config$ make --debug MACHINEROOT=../machine/quanta MACHINE=quanta_common_rglbmc all
GNU Make 4.3
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Reading makefiles...
Updating makefiles....
Updating goal targets....
 File 'all' does not exist.
   File '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/kernel-source' does not exist.
     File '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/tree' does not exist.
       File '/home/onie/onie/build/stamp-project' does not exist.
      Must remake target '/home/onie/onie/build/stamp-project'.
mkdir: created directory '/home/onie/onie/build/images'
mkdir: created directory '/home/onie/onie/build/download'
      Successfully remade target file '/home/onie/onie/build/stamp-project'.
    Must remake target '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/tree'.
mkdir: created directory '/home/onie/onie/build/quanta_common_rglbmc-r0'
mkdir: created directory '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp'
mkdir: created directory '/home/onie/onie/build/quanta_common_rglbmc-r0/initramfs'
    Successfully remade target file '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/tree'.
     File '/home/onie/onie/build/download/kernel-4.9.95-download' does not exist.
    Must remake target '/home/onie/onie/build/download/kernel-4.9.95-download'.
==== Getting Linux ====
  Downloading linux-4.9.95.tar.xz from: https://www.kernel.org/pub/linux/kernel/v4.x/linux-4.9.95.tar.xz
2023-03-31 12:40:29 URL:https://mirrors.edge.kernel.org/pub/linux/kernel/v4.x/linux-4.9.95.tar.xz [93289700/93289700] -> "/home/onie/onie/build/download/linux-4.9.95.tar.xz" [1]
  Got          linux-4.9.95.tar.xz

linux-4.9.95.tar.xz: OK
    Successfully remade target file '/home/onie/onie/build/download/kernel-4.9.95-download'.
  Must remake target '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/kernel-source'.
==== Extracting Linux ====
  Successfully remade target file '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/kernel-source'.
   File '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/kernel-patch' does not exist.
  Must remake target '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/kernel-patch'.
==== patching  Linux ====
hint: Using 'master' as the name for the initial branch. This default branch name
hint: is subject to change. To configure the initial branch name to use in all
hint: of your new repositories, which will suppress this warning, call:
hint:
hint:   git config --global init.defaultBranch <name>
hint:
hint: Names commonly chosen instead of 'master' are 'main', 'trunk' and
hint: 'development'. The just-created branch can be renamed via this command:
hint:
hint:   git branch -m <name>
Initialized empty Git repository in /home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux-4.9.95/.git/
Auto packing the repository in background for optimum performance.
See "git help gc" for manual housekeeping.
Checking for changes in the working directory ... done
No patches applied
  Successfully remade target file '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/kernel-patch'.
   File '/home/onie/onie/build/quanta_common_rglbmc-r0/stamp/kernel-build' does not exist.
     File '/home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux/.config' does not exist.
    Must remake target '/home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux/.config'.
==== Copying conf/kernel/4.9.95/linux.x86_64.config to /home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux/.config ====
'conf/kernel/4.9.95/linux.x86_64.config' -> '/home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux/.config'
==== Merging patches from /home/onie/onie/machine/quanta/quanta_common_rglbmc/kernel/config to /home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux/.config ====
Using /home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux/.config as base
Merging /home/onie/onie/machine/quanta/quanta_common_rglbmc/kernel/config
Value of CONFIG_MODULES is redundant by fragment /home/onie/onie/machine/quanta/quanta_common_rglbmc/kernel/config:
Value of CONFIG_I2C_I801 is redefined by fragment /home/onie/onie/machine/quanta/quanta_common_rglbmc/kernel/config:
Previous value: # CONFIG_I2C_I801 is not set
New value: CONFIG_I2C_I801=y

Value of CONFIG_I2C_ISCH is redundant by fragment /home/onie/onie/machine/quanta/quanta_common_rglbmc/kernel/config:
Value of CONFIG_I2C_PIIX4 is redefined by fragment /home/onie/onie/machine/quanta/quanta_common_rglbmc/kernel/config:
Previous value: # CONFIG_I2C_PIIX4 is not set
New value: CONFIG_I2C_PIIX4=y

Value of CONFIG_IRQ_DOMAIN_DEBUG is redundant by fragment /home/onie/onie/machine/quanta/quanta_common_rglbmc/kernel/config:
Value of CONFIG_IPMI_HANDLER is redefined by fragment /home/onie/onie/machine/quanta/quanta_common_rglbmc/kernel/config:
Previous value: # CONFIG_IPMI_HANDLER is not set
New value: CONFIG_IPMI_HANDLER=y

#
# merged configuration written to /home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux-4.9.95/.config (needs make)
#
    Successfully remade target file '/home/onie/onie/build/quanta_common_rglbmc-r0/kernel/linux/.config'.
     File '/home/onie/onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.38/stamp/xtools-build' does not exist.
       File '/home/onie/onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.38/build/.config' does not exist.
         File 'conf/crosstool/gcc-6.3.0/uClibc-ng-1.0.38/crosstool.x86_64.config' does not exist.
        Must remake target 'conf/crosstool/gcc-6.3.0/uClibc-ng-1.0.38/crosstool.x86_64.config'.
make: *** No rule to make target 'conf/crosstool/gcc-6.3.0/uClibc-ng-1.0.38/crosstool.x86_64.config', needed by '/home/onie/onie/build/x-tools/x86_64-g6.3.0-lnx4.9.95-uClibc-ng-1.0.38/build/.config'.  Stop.

I used a Debian 10 vm, Debian 11 vm and also a DUE env but the error is the same.
I think it's related to GCC 6.3.0 compiler because it's not happening with older versions of compiler.
Could you please help me to solve it?
Many Thanks
Best regards

luigitalboy, 2023-03-31:
@mshych  could you help me? Thanks

mshych, 2023-04-03:
Hi,
I think that https://ocp-all.groups.io/g/OCP-ONIE/topics is a more suitable place for such a question.
According to  https://github.com/opencomputeproject/onie/blob/master/build-config/scripts/onie-build-targets.json#L241 ,
platform quanta_common_rglbmc has an old build environment: Debian 9 and 2020.05br release branch.
Please try to build in this environment.

luigitalboy, 2023-04-04:
Thanks a lot @mshych ! :-)