opencomputeproject/onie #1053: error on make -j4 MACHINEROOT=../machine/celestica MACHINE=cel_seastone all

mrafatpanah, 2024-03-02:
I am trying to build an image for cel_seastone using the due-onie-build-debian-9 environment. Initially, I encountered the following error:

`make: *** No rule to make target 'conf/crosstool/gcc-4.9.2/uClibc-ng-1.0.38/crosstool.x86_64.config', needed by '/home/rafet/Aryan/patch_test/onie/build-cell/x-tools/x86_64-g4.9.2-lnx3.2.69-uClibc-ng-1.0.38/build/.config'. Stop.
`
Can anyone help me understand why this is happening?"

chacal1231, 2024-06-24:
Have same problem, you solve that?

robinchrist, 2024-07-13:
Versions newer than 2021.08 do not seem to build. Celestica probably hasn't updated their stuff for the breaking changes in 2021.11, see here https://opencomputeproject.github.io/onie/developers/building.html

I have written up some stuff on how to build for DX010: https://gist.github.com/robinchrist/81edaf15143f57722aa34515ff037231

chacal1231, 2024-07-15:
> Versions newer than 2021.08 do not seem to build. Celestica probably hasn't updated their stuff for the breaking changes in 2021.11, see here [...]

Man, will try!

robinchrist, 2024-07-15:
> Man, will try!

I have also created a repo here: https://github.com/robinchrist/ONIE-Adventures

which contains build binaries..

chacal1231, 2024-07-16:
> > Man, will try!
> 
> I have also created a repo here: https://github.com/robinchrist/ONIE-Adventures
> 
> which contains build binaries..

You're the man! thanks a lot!